open-ephys/plugin-GUI #446: Plugin with external dependencies not working under Windows 10

rhayman, 2021-05-11:
Hi,

I have written a plugin (https://github.com/rhayman/PosTracker/tree/development) that has an external dependency on openCV. I recently re-wrote / re-structured the plugin to be consistent with the new open-ephys plugin format. I have been able to successfully compile and use the plugin under Linux (Ubuntu 18.04) but have not been able to do so under Windows 10.

The error output I get in the Debug console reads:

`C:\Users\Robin\Documents\plugin-GUI\Source\Processors\PluginManager\PluginManager.cpp:219: Failed to load plugin DLL: DLL Error 0x7e`

The point of failure is with the call to LoadLibrary in the PluginManager class. The output from tools like dumpbin appear to show that the dll that results from building the plugin under Windows is linking correctly to openCVs libraries.

I am using version 0.5.4 of the plugin-GUI, Visual Studio 16 (2019), CMake 3.20.2 and the last of my patience with Windows. Unfortunately, I need to use Windows to use the Neuropixels probes...

It's entirely possible I'm missing something obvious - any help / suggestions / advice would be most welcome.

jsiegle, 2021-05-13:
Hi Robin – we will figure out how to get this plugin working on Windows; our best guess is that there's a problem with the way your CMake file is set up, but we'll have a more definitive answer soon.

FWIW, we have successfully acquired Neuropixels data on Linux. However, the Enclustra Linux driver is quite finicky, and can sometimes cause the entire machine to freeze up. We are communicating with Enclustra about getting this fixed. Once that's sorted out, we'll be able to officially release the plugin for Linux, but unfortunately we don't know exactly when that will happen.

anjaldoshi, 2021-05-14:
Hi Robin,

I was able to build and load the plugin successfully on my machine. I used the code from the "[windows](https://github.com/rhayman/PosTracker/tree/windows)" branch of your repo. I used OpenCV v4.5.2 which I built from source along with the contrib modules. 

All I did was modify the CMakeLists to refer to the correct location of OpenCV libs and DLLs. I removed all the OpenCV related files that were present in the repo, and added the files that I built from source and rearranged them so that Cmake can easily find all the required files. I also added a line in CMakeLists which installs all the required OpenCV DLLs in the shared directory of  plugin-GUI which are required by PosTracker during runtime. This fixes the `Failed to load plugin DLL` error that you were facing.

With my changes, you don't even have to install OpenCV anymore. The plugin uses the OpenCV files shipped in the repo. The  disadvantage of this is that the plugin is stuck with using the current latest version of OpenCV.

I've created a [PR](https://github.com/rhayman/PosTracker/pull/4) on your repo with all of my changes in it. Take a look at it and try it out. 

P.S. I wasn't able to test whether all the functionalities of your plugin works or not with my changes, so you'll have to make sure of that.

rhayman, 2021-05-17:
Hey Anjal / Josh,

Thanks so much for the PR and I'm glad you got things working but I'm sorry to say I'm still struggling here. TL;DR I'm still getting the same error as before.

I see you added a line to CMakeLists which copies the dll's from the PosTracker/libs/windows/x64/vc16/bin directory to the /shared directory of the plugin-GUI build during the install step. However those files weren't added to the PR/ repo probably because there is a line in the repos .gitignore that ignores bin/ directories. Following the work you did though I copied the directories / files from my openCV (4.5.2) build directory to the PosTracker/libs/windows/ directory manually i.e. including the directories "include", "x64" and the openCV cmake files. The build and install went fine but upon running open-ephys.exe I still get the same `Failed to load plugin DLL`. Not that I think it makes much difference but I'm getting this against both Release and Debug build types.

In desperation, I've tried setting/ unsetting various environment variables (additions to $PATH) with no luck. I've done nothing extra with my openCV build beyond compile it in it's own build directory. I'm also assuming there's no special Visual Studio magic I need to invoke such as telling it where various files / directories are? I assumed that CMake would provide it info to do with dependencies, library locations and so on.

Thanks again @anjaldoshi for spending the time to look at this. @jsiegle - good to hear about the forward movement with the Enclustra Linux driver, I look forward to the day I don't have to boot into my Windows partition

anjaldoshi, 2021-05-17:
Hi Robin,

I've created a new [PR ](https://github.com/rhayman/PosTracker/pull/5)that updates the `.gitignore` and adds the bin files back. Can you test it and see if that fixes the issue? 

If it still doesn't, can you send me the output of when you run CMake to generate the build files?

And yes, you only need to run CMake and the build the plugin with Visual Studio without making any changes in Visual Studio except selecting the correct Build (Release/Debug) type.

rhayman, 2021-05-18:
Hi Anjal,

Thanks for the PR - I'm very happy to report it fixed the issue and I can now load and use the plugin :)

There are some minor issues with the controls for the camera under Windows vs Linux but that is entirely down to my rushed coding.

Thank you again for your time and help with this.

closed 2021-05-18 by rhayman