open-ephys/plugin-GUI #447: Digital in not showing up

Gittinator, 2021-05-14:
I can't see the digital in from my Intan USB board. I know they're supposed to show up in the "shaded region" of the LFP Viewer, but I can't figure out where that is. The data channels and ADC show up fine.

jsiegle, 2021-05-14:
Hi! Can you provide some more info about your setup? Operating system, GUI version, and a description/image of your signal chain would be helpful.

Gittinator, 2021-05-14:
GUI version is 0.5.4, I've tried on Windows and OS X. The signal chain is just the rhythm FPGA and LFP Viewer

jsiegle, 2021-05-14:
Can you send a pic of the hardware configuration? What's the digital signal you're using for testing?

Gittinator, 2021-05-14:
Yes, what would you want the picture of? It's just the Intan USB board with a signal from a rotary encoder plugged into a digital input pin. It shows up as expected on the Intan software. 

My main question is just how to see the digital inputs, which aren't showing up at all, it's not that the signal isn't showing up. The documentation says that they should be in the "shaded region" of the LFP Viewer. I can see the ADCs, data, and AUX inputs.

jsiegle, 2021-05-14:
The event overlay may not appear if the digital inputs are changing too quickly. It will also only display changes in the digital inputs (i.e., when the encoder is moving), not the starting value.

Do the encoder events show up in the recorded data? You can test this by dropping a Record Node in between the Rhythm FPGA and LFP Viewer, and checking whether the event data files increase in size.

jsiegle, 2022-07-30:
If this is still a problem in the latest version of the GUI, please re-open this issue.

closed 2022-07-30 by jsiegle