open-ephys/plugin-GUI #448: Neuropix-PXI plugin overflow

rhayman, 2021-05-21:
A simple signal chain consisting of the plugins Neuropix-PXI -> Probe Viewer is resulting in an overflow error after about 25-30 seconds following the start of a recording. The exact error message is:

`DataQueue::writeChannel Recording Data Queue Overflow: sz1: 0 sz2: 0 nSamples 1024`

The nSamples value can change.

The CPU load status bar spends a lot of its time at 100% (some times going down briefly). I can still record with this happening and the resulting data looks fine.

This is on a Windows 10 machine, v 0.5.4 of the plugin-GUI and the latest Neuropix-PXI plugin available from the plugin manager.

jsiegle, 2021-05-21:
Are you recording to an SSD? And does the CPU load only go up once you start recording, or is it high as soon as you start acquisition?

rhayman, 2021-05-21:
My bad - it's an HDD, I thought I was recording to one of the machines
SSD's. FYI the error popped up after about 30 seconds.

closed 2021-05-21 by rhayman

jsiegle, 2021-05-21:
Yeah, that's consistent with our experience recording Neuropixels data to an HDD. It should run smoothly with an SSD.