open-ephys/plugin-GUI #449: trouble installing nidaqmx plugin

ffredes, 2021-05-24:
Hi there, I cannot install the NIDAQmx,  and spectrum-viewer plugins. Getting the following error:
D:\a\plugin-GUI\plugin-GUI\Source\Processors\PluginManager\PluginManager.cpp:219: Failed to load plugin DLL: DLL Error 0x7e
Matlab interface and other plugins installed properly. 
I have tried to uninstall and reinstall visual studio community 2019 but the problem persists.  
I have seen a similar issue in #446. 
I am using the last version of the GUI. 

Any idea? 
Thank you!

anjaldoshi, 2021-05-24:
Hi,

For the NIDAQmx plugin, can you make sure you installed the drivers required for it to work? They can be downloaded from here: https://www.ni.com/en-us/support/downloads/drivers/download.ni-daqmx.html

The reason behind Spectrum Viewer plugin not working is that we forgot to add OpenEphysFFTW as its dependency in Plugin Installer's upstream cloud service. I just fixed that. To make the plugin work, you'll have to follow these steps:
- close the GUI, if it is running
- delete the `specturm-viewer.dll` file from the GUI's `plugins` directory
- run the GUI
- open Plugin Installer
- install the Spectrum Viewer plugin again 

That should fix it.

jsiegle, 2022-07-30:
As Anjal said, the NIDAQmx plugin does require the appropriate drivers in order to work correctly.

The Spectrum Viewer plugin installation should be fixed in GUI v0.5.X; we haven't migrated it over to v0.6.X yet, but are planning to do that soon.

closed 2022-07-30 by jsiegle