open-ephys/plugin-GUI #450: Open ephys gui v0.5.4 error with DIN (not occuring in v0.5.3.1)

timsainb, 2021-05-26:
I tried upgrading to the new version of the GUI but I'm getting a new error related to digital input. It seems like the digital input on the open ephys box, with the new gui, is keeping my raspberry pi from being able to flip an I2C switch. 

I'm wondering if there's anything new going on in the backend that might cause something like that? 

For now, I've just downgraded back to v0.5.3.1

Basically, I've got a raspberry pi that sounds a signal about light information through the DIN of open ephys. With GUI v0.5.4, I get this error message on the raspberry pi when trying to flip that switch:

```
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 2737, in i2c_write_byte_data
    self.sl, _PI_CMD_I2CWB, handle, reg, 4, extents))
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 975, in _u2i
    raise error(error_text(v))
pigpio.error: 'I2C write failed'
```

Here's what happened when I upgraded to the new GUI:
- I2C error: when open ephys is then turned on and everything is plugged in
- no I2C error: when the digital input (DIN) cable is unplugged from the open ephys box
- no I2C error:  when the DIN cable is plugged into the open ephys box, but the open ephys box is not plugged into the computer
- no I2C error: when DIN is plugged in, cable is plugged in, and open ephys gui is off
- no I2C error: with previous version of GUI

jsiegle, 2021-05-26:
@aacuevas ...do you any idea about what could be causing this? My guess is that it's related to this change: https://github.com/open-ephys/plugin-GUI/pull/419

If so, then it's most likely a bitfile issue, and you can swap out the old `rhd2000_usb3.bit` with the new one to fix the problem, while still using the latest GUI version.

@timsainb, how is your Raspberry PI connected to the digital in breakout board? That may help diagnose the problem.

timsainb, 2021-05-26:
We're using a custom PCB with a PCA9685 PWM LED Driver that interfaces with the i2c channel on raspberry pi (ch 55). I believe the I2C channel is routed directly to a connector, which then plugs right into the raspberry pi. The same is true of the other, non i2c, digital channels. @theilmbh could probably provide more info.

aacuevas, 2021-05-27:
Hi,
As Josh said, we would need to know how is it connected to the Open Ephys board.

Just as an idea, is your i2c bus, by any chance, connected to the lines labeled "i2c" in the IO board?
Those are not actually i2c and should not be used in any i2c bus. We had the idea of maybe using them in the future and so we reserved them, but never found a proper use case. In previous versions they were left with high impedance, so they were basically unconnected and unused, but in #419 we decided to use them for HARP timing synchronization.

jsiegle, 2022-07-30:
It sounds like this is likely a hardware issue. If you're still having this problem, please re-open this issue and provide more details about how your PCB is connected to the acquisition board digital inputs.

closed 2022-07-30 by jsiegle