open-ephys/plugin-GUI #452: Plugins installation not going through on Windows 10

Barryjuait, 2021-05-27:
Hi,
I have been trying several times, but the plugins don't get installed.

This is what I do:
- Fetch development branch of the plugin-GUI repository.
- I create the Build files with cmake (Visual Studio 12 2013 Win64).
- I open "open-ephys-GUI.sln" (in the installation guide says that it is called "open-ephys.sln", but I don't think that is a problem) with Visual Studio 2013.
- I Select Debug x64 and build the ALL_BUILD file. When I hit build it warns me that all the solution are outdated, but I run it anyway.
- The whole process finishes with 1 success (Zero Check), and 17 fails, so I cannot use the rest of the plugins.

Help please!!

Sebastian

anjaldoshi, 2021-05-27:
Hi Sebastian,

We advice using the current latest version of Visual Studio (VS 16 2019), but it should still work with VS2013. I'll install VS2013 on my machine and see if I encounter any errors. Meanwhile, can you try to build the GUI with VS2019? That should be the easiest fix for now. Make sure you run create the Build files again for VS2019 as instructed over [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-the-GUI.html).

Also, is there a reason you're building the GUI in Debug mode? If you're not trying to debug any errors/issues, then I suggest using the Release build type. Release mode build is optimized for production use so it should have much better performance than Debug mode.

Barryjuait, 2021-05-27:
Thanks @anjaldoshi. I will try with VS2019.

I did compiled in Release Mode as well, and I got the same outcome :(

Barryjuait, 2021-05-28:
Hi @anjaldoshi, that was a big difference! Almost all was successfully compiled with VS2019, except for the spike sorter plugin.

18>Done building project "SpikeSorter.vcxproj".
19>------ Skipped Build: Project: ALL_BUILD, Configuration: Debug x64 ------
19>Project not selected to build for this solution configuration 
========== Build: 18 succeeded, 0 failed, 0 up-to-date, 1 skipped ==========

Barryjuait, 2021-05-28:
Update: it is working fine, but it crashes when it reloads a configuration that includes ADC or AUX channels. If they are unselected the GUI loads everything correctly.

jsiegle, 2021-05-28:
If you're running in debug mode, can you see the line where it's crashing on reload?

anjaldoshi, 2021-05-28:
> `18>Done building project "SpikeSorter.vcxproj".
19>------ Skipped Build: Project: ALL_BUILD, Configuration: Debug x64 ------
19>Project not selected to build for this solution configuration
========== Build: 18 succeeded, 0 failed, 0 up-to-date, 1 skipped ==========`

Looks like the SpikeSorter plugin project got skipped. Can you make sure that Project is enabled in the solution in Visual Studio.

> it crashes when it reloads a configuration that includes ADC or AUX channels. If they are unselected the GUI loads everything correctly.

Can you send us the signal chain you are using?

Barryjuait, 2021-05-31:
@anjaldoshi I think the Project it is enabled. This is how it looks in VS:

[image]

Barryjuait, 2021-05-31:
@jsiegle I ran now the program with a configuration having ADC and AUX channels enabled. The program crashed in line 238:

[image]

jsiegle, 2021-05-31:
I'm not able to replicate this crash. Are you sure you have the latest version of the code? This was definitely a problem with an earlier release, but it should be fixed now. Also, the LfpDisplayNodeBeta was deprecated in v0.5.4, so that shouldn't be showing up.

Barryjuait, 2021-06-02:
@jsiegle You are super right. Now I verified that I had the v0.5.3. Maybe I'm lost in some very basic step. This is what I do:
I cloned the repository in my computer through Github Desktop, and then I selected the development branch (not the upstream/development) and I hit "Fetch origin". I would expect that each time I fetch it I get the new version?

jsiegle, 2021-06-02:
I think you do need to select `upstream/development` in this case, because that points to the git repository owned by `open-ephys`, whereas (I'm assuming) `development` points to your repository, which will have whatever code was present when you forked it.

It's also worth noting that the `upstream/master` branch will always be consistent with the latest release, whereas `upstream/development` may have new changes that have not been fully tested yet.

jsiegle, 2022-07-30:
Closing this issue, as it seems to be resolved. Note that the latest version of the GUI now lives on the `main` branch.

closed 2022-07-30 by jsiegle