open-ephys/plugin-GUI #453: SMA port triggered aquisition crashes GUI (ZMQ module installation crashes GUI)

catubc, 2021-06-01:
Hello
We are using OpenEhys for neuropixels system. Version: v0.5.3.1.  When we set Trigger=1 and start the recording to wait for an external trigger via the SMA port, the GUI crashes.  If we just visualize/play through the LFP viewer the SMA trigger is visible (as yellow shading).

To resolve this on an older version of OpenEphys we used the ZMQ interface to trigger the acquisition. But now both the 0.1.0 and 0.2.0 AP17 plugin cause the GUI to crash on install.

Any suggestions?  What are the suggested methods for triggering the system.

jsiegle, 2021-06-01:
Hi Cat!

I assume the first issue you're having is with the "Record Control" plugin? Can you describe the full signal chain you're using, or send the settings XML file? We'll try to replicate it on our end.

For the second issue, we'll look into why the ZmqInterface is not installing correctly. However, it's actually the "NetworkEvents" plugin that's used to trigger acquisition/recording. Can you install that one? If so, you can use either [open-ephys-matlab-tools](https://github.com/open-ephys/open-ephys-matlab-tools) or [open-ephys-python-tools](https://github.com/open-ephys/open-ephys-python-tools) to control the GUI.

jsiegle, 2022-07-30:
If you're still experiencing these problems with the latest version of the GUI (0.6.X), please re-open this issue.

closed 2022-07-30 by jsiegle