open-ephys/plugin-GUI #454: Spike Viewer time window is off-center

Barryjuait, 2021-06-02:
Hi,
It looks like the time window of the detected spikes are totally off toward the right, so I miss the whole depolarization period. Im using GUI v.0.5.4 

[image]

Barryjuait, 2021-06-02:
Update: the problem is solved if I use Spike sorter in the pipeline instead of Spike detector before the spike viewer.

[image]

jsiegle, 2021-06-02:
Yeah, the spike polarity conventions on the Spike Detector need to be updated; please use the Spike Sorter for now.

jsiegle, 2022-07-30:
This issue has been fixed in the Spike Detector for a while now. Also, in the latest version of the GUI, only the Spike Detector can detect spikes (the Spike Sorter just adds a `sortedID` label).

closed 2022-07-30 by jsiegle