open-ephys/plugin-GUI #455: Synching a 5V pulse with Neuropixels data stream

Darkweiss, 2021-06-04:
We have a function generator sending a 5V pulse to a NI PXIe-6341 through a BNC. The pulse shows up on the NIDAQMX data source (picture). We are also able to detect the rising and falling edge events with crossing detector. However, so far we have been unable to synchronise these TTL events with the data from a single Neuropixels probe in order to plot PSTH plots. The PSTH plots below were made with spike times from Kilosort2  and the events timestamps were calculated as TTL event minus the time of the first time sample of NIDAQMX. However, the first time sample of NIDAQMX is not the same as the Neuropixels one. What is the easiest way to synchronise these 2 data streams?
[image]
[image]

medengineer, 2021-06-04:
You are very close! In order to synchronize, the 5V pulse needs to feed into a _digital_ input of the NI card, whereas in your example here, the right LFPViewer shows the 5V pulse is feeding into the first _analog_ input of the card.

Darkweiss, 2021-06-07:
That now works. Thank you @medengineer! However, when looking at the event times of the crossing detector plugin, they do not seem to be aligned with the pulse. I guess we can do thresholding from the analogue signal but it would be nice if we already have the event times from Open Ephys.
[image]

medengineer, 2021-06-07:
The CrossingDetector events are not exactly aligned with the Neuropixels events in the LFPViewer because synchronization only occurs in the RecordNode and the synchronized data only propagates to the recorded files (and not downstream processors). In the recorded data, the CrossingDetector event timestamps will each map to a synchronized source processor timestamp (in seconds) and the events should be aligned as expected.

closed 2021-06-08 by Darkweiss