open-ephys/plugin-GUI #457: Error in timestamps for buffers with more than 1024 points

bmsousa91, 2021-06-14:
I'm using the Open Ephys, together with Pulse Pal, to record channels from an Intan RHD 2132 headstage and to trigger closed-loop stimulation. The GUI version is v0.5.4. I noticed some jitters on timestamp data when testing different buffer sizes. Only buffers with more than 1024 points present this issue and it gets progressively worse as the buffer size increases. I'm using "load_open_ephys_data_faster.m" to load the data and the info.ts variable is not correctly spaced by 1024.

For the figure attached I recorded only the first channel using the following process chain: FPGA Rythm (30 kHz) -> Bandpass Filter -> Record Node -> LFP Viewer. I'm aware of the announcement made about timestamp issues (https://github.com/open-ephys/data_fix_0.4.2) and I tested the following actions, with no success: i) marked the "use separate files" recording option; ii) clicked first on the recording button, as opposed to clicking first in the play button and then record; iii) recorded more than 1 channel (channels 1, 2 and 3).

Would you know if there is something wrong with my settings? Thanks in advance!

[image]

medengineer, 2021-06-14:
Thanks for reporting this, your settings are correct. We are able to replicate this issue on our end and are working on a fix.

jsiegle, 2021-07-02:
We figured out the source of this problem (the Record Node `DataQueue` was getting overrun if more than 1024 samples arrived in a single buffer). It's fixed in pull request #460, which we'll merge before the next release.

bmsousa91, 2021-07-02:
Great! Thank you for the support.

closed 2022-07-30 by jsiegle