open-ephys/plugin-GUI #458: No Neuropixels-PXI sync info in sync_messages.txt

hwpdalgleish, 2021-06-15:
Hi,

I'm recording from a single-shank Neuropixels 2.0 probe using OpenEphys v0.5.3.1 (compiled from source) and I am finding that the recording start time and acquisition rate is not being recorded in the relevant sync_messages.txt file. Is this expected for 2.0 probes and/or this version of Open Ephys?

For reference, my signal chain is:
_Neuropix-PXI -> Common Avg Reference -> Record Node 102 -> BandPass -> LFPViewer_

It was my understanding that the recording start time and sampling rate should be recorded in:
_exptName/Record Node 102/experiment1/recording1/sync_messages.txt_

In the manner below:
_Processor: Neuropix-PXI Id: 101 subProcessor: 0 start time: 1@30000Hz_

As I've seen previously with older versions of Open Ephys and 1.0 probes. However, as mentioned above, this file is empty. This is not the case if I record from a Rhythm FPGA plugin with this version of Open Ephys. In this case I find this in the relevant sync_messages.txt file:
_Processor: Rhythm FPGA Id: 110 subProcessor: 0 start time: 3042560@30000Hz_

Thanks for your help!

Best,
Henry

medengineer, 2021-06-15:
Hi Henry,

This bug was fixed in the 0.5.4 release of the GUI which can be found [here](https://github.com/open-ephys/plugin-GUI/releases).

jsiegle, 2021-06-15:
I would also add that the start time is redundant with the first recorded timestamp, so it can be recovered if it's missing from sync_messages.txt

closed 2022-07-30 by jsiegle