open-ephys/plugin-GUI #462: RHD2132 headstage not detected by the GUI (Windows)

chiyu1203, 2021-07-18:
Hi my question is quite simliar to #219 but I used Windows 10 so I thought I made another question here.
I have been using RHD2132 headstage with openEphys board v2.2 to do recording (GUI version 0.4.5) for the past two years. It was fine at least until my last recording in this Feb 2021.
This week, when I run GUI, our USB3 open-ephys board and the ADC channel were detected, but the RHD2132 headstage were not. Hitting rescan did not help. Swapping with another openEphys board or headstage didn't help either. So I reckon it was a software issue as the Windows kept asking me to update some softwares and a new security app popped out (Microsoft Defender SmartScreen). I went to the debug console but dont know what to do with it
[image]
[image]
[image]
[image]
[image]

Could anyone suggest me what I should do?

Best Regards
Chi-Yu

chiyu1203, 2021-07-19:
Hi everyone, sorry we fixed the issue. It was because of the headstage cable. We changed to another headstage cable and it worked now. Does anyone know where we can buy a new headstage cable (we already have few starter kits so just want to buy a cable)?

jsiegle, 2021-07-19:
Glad you solved it!

You can buy individual cables from Intan, in either the standard or ultra-thin widths: https://intantech.com/pricing.html

closed 2021-07-19 by jsiegle