open-ephys/plugin-GUI #463: Random freeze in LFP viewer and recording error "Error in Rhd2000EvalBoard::readDataBlock: Incorrect header. "

bouromain, 2021-07-26:
Hi,

I have a recurring problem that happens randomly in my recordings while I am recording. Most of my recordings do not encounter any problems but for some, the LFP viewer freezes, while other components or the GUI stay active (recording time, interaction with start stop button,...). During this crash the console returns this error code:

`Error in Rhd2000EvalBoard::readDataBlock: Incorrect header. `

This error has happened with:
- this can happen after 2 to 60min recordings
- no other program running on the computer and no other usb port in use (other than keyboard and mouse)
- two different boards, different USB cables and on different USB ports
- recording on a sata SSD
- with versions 0.4.6 0.4.5 and 0.5.4 (on Ubuntu compiled from source)
- with ADC board plugged on analog input (I also record 3 ADC channels)
- recording of 32 channels and ADC (20kHz)
- a really simple signal chain (Rhythm FPGA-> LFP viewer-> recording node)

This error seem to be very similar [this one reported in the google group](https://groups.google.com/g/open-ephys/c/0n1rCZaCLr8)

I thought it was specific to my setup but today a colleague  encountered a similar one using windows (version 0.5.4 compiled from source) and a neuropixel probe.

Thank you in advance for your help.

Best

Romain

jsiegle, 2021-07-26:
This sounds like an error in USB data transmission. Is the acquisition board connected to a USB 3.0 port? What are the specs of your acquisition computer?

Can you ask your colleague using Neuropixels to open a separate issue? Even though the symptoms are similar, it most likely has a different underlying cause.

bouromain, 2021-07-27:
Hi Josh,

Thank you for your quick answer. 

Yes, the board is connected to a usb3 port. 
The computer has a i7 (4 cores) 16go of ram and SATA SSD. 

I'll ask my colleague to report a specific issue.

jsiegle, 2021-07-27:
Thanks...I can't think of anything obvious that would explain the error, but maybe there's something more subtle going on.

A few more questions:
- Where did you get your acquisition board from?
- What version of Ubuntu are you using?
- Have you tried using the pre-compiled version of the GUI, or have you only built it from source?

bouromain, 2021-08-02:
I am not sure about the origin of the board. But I had the problem with two different boards, so it migth be unlikely that this is the source of the problem.

I am currently using Ubuntu 18.04 but I had the problem with 16.04. 

I think I never tried using pre-compiled source as I need a particular plugin for my recordings. 

Do you think that the problem is more likely to be a reading (reading data from the board, usb port,...) or writing problem (slow writing to disk,...)? Is there a way to make this error more informative to identify the source of the problem?

thank you again for your answers.

jsiegle, 2021-08-02:
It's almost certainly a reading problem.

@aacuevas do you have any thoughts on this? Could they try changing the size of the USB data buffers?

aacuevas, 2021-08-02:
Are you buliding the sources in Debug or Release mode? The cmake script creates a debug executable by default, unless you manually set the appropriate option. I've seen this kind of read issues when using a linux debug build.

aacuevas, 2021-08-05:
It should not happen in release mode indeed. There is something definitely going wrong in USB transmission. 
Can you send me to aaron (at) open-ephys.org the full console contents (up to acquisition start. Unless there is any other message from acquisition start until the errors you mentioned before) and the contents of `lsusb -t` and `lsusb -v` ?

jsiegle, 2022-07-30:
The plugin for interfacing with the acquisition board now lives here: https://github.com/open-ephys-plugins/rhythm-plugins

If you're still experiencing this problem with the latest version of the GUI, please open an issue in that repository.

closed 2022-07-30 by jsiegle