open-ephys/plugin-GUI #464: ValueError: could not broadcast input array from shape (320956416,) into shape (320958464,)

barbaracoimbra, 2021-07-28:
Hello,

I am fairly new to using OpenEphys and I've been performing recordings in the past months. So far, everything is going ok, however I encountered a problem in one of my recording sessions and I cannot convert my .continuous data to .dat and I get the following error:

Traceback (most recent call last):
  File "python_ephys_analysis.py", line 63, in <module>
    oe.pack_2(folder, filename = '102_CH.dat', channels = channels, chprefix = 'CH', dref = None, session = '', source = '102')
  File "D:\Shared_Ephys\Barbara\open-ephys-python-tools-main\Python3\OpenEphys.py", line 431, in pack_2
    data_array = loadFolderToArray(folderpath, channels, chprefix, np.int16, session, source)
  File "D:\Shared_Ephys\Barbara\open-ephys-python-tools-main\Python3\OpenEphys.py", line 100, in loadFolderToArray
    data_array[:, i + 1] = loadContinuous(os.path.join(folderpath, f), dtype)['data']
ValueError: could not broadcast input array from shape (320956416,) into shape (320958464,)

Can anyone please help me?

jsiegle, 2021-07-28:
It looks like you're concatenating a bunch of continuous data into a large array, but the number of samples in one file is different than the other files. If you can give a few more details about what you're trying to do, as well as some information about your recording, I can try to point you in the right direction.

jsiegle, 2022-07-30:
If you're still having trouble with this, please open a new issue in the [`open-ephys-python-tools`](https://github.com/open-ephys/open-ephys-python-tools) repo.

closed 2022-07-30 by jsiegle