open-ephys/plugin-GUI #465: Dropped data from recording

Hankslab-ABG, 2021-07-28:
We had a set of TTL pulses (totaling approx 16ms) that never showed up in our recording. These pulses denoted the start of a behavioral trial (pulse 1) followed by a 15-bit binary readout of the trial number. Trial 95 showed up fine, trial 96's TTL pulses were missing entirely, and trial 97's showed up fine.

Oddly, when we went back and examined the trial start timing the time between the start of trial 95 and the start of trial 97 was less than the amount of time that should have elapsed when we added the total trial length of trials 95 and 96 (both of which registered in the behavior readout from our Bpod).

I looked into it and there are indeed gaps in the continuous timestamps.npy file that perfectly overlap with when the missed behavioral trial's TTL pulses should have been transmitted.

The odd thing, though, is that the number of samples in the .dat file weren't changed. Would we expect them to if there was dropped data? The number of samples match once the gaps are taken into account, that is: timestamps(end)-timestamps(1)+1=number of samples in continuous.dat.

In speaking with @jsiegle via the neuropixels slack channel he indicated that "There should be zeros in the continuous file at the samples where the timestamps jump".

I'm currently checking into that now.

@jsiegle also indicated I should post this issue to the github for tracking.

jsiegle, 2021-07-29:
Thanks for posting. This sounds like the data buffers for one or more processing cycles were not written to disk by the Record Node. I don't think this is something we've seen before, but we'll do our best to track down the cause.

Can you send the settings.xml file from this recording? Also, was there any indication that your CPU load was higher than usual (based on the CPU monitor in the upper left of the GUI window)?

Hankslab-ABG, 2021-07-29:
No problem.

It seems I can't post the xml file here. Is there an email address I can send it to?

Honestly we weren't paying much attention to the CPU load during the recording. Mostly we're watching the video feed to make sure the rat isn't unplugging itself or getting its tether tangled up. I can't say for sure we didn't have chrome open or Matlab open, but it's unlikely that we were doing anything like running Kilosort in parallel. It doesn't sound like background processes like windows update, antivirus, etc would be spiking the CPU load enough to cause an issue, is that correct?

I've checked to see if the samples in the continuous binary file are all zeros where the gaps occur. Oddly they are not.

For the first gap the continuous file does indeed read all zeros across all channels (10768 samples total). For the 2nd gap (28161 samples total) it reads all zeros for the first chunk of the gap (samples 1:6671) then has non-zero values for samples 6672:10767 and then has zero values again for samples 10768:28161.

Is there a scenario you could imagine where there was a gap in the timestamps.npy file but non-zero values in the binary file?

Hankslab-ABG, 2021-07-29:
Just to make sure I'm not examining this erroneously, below is the Matlab code I'm running to read out the signal for each gap:

gaps=find(diff(AP_cont_timestamps)>1);
tstart_first_gap=AP_cont_timestamps(gaps(1));
tend_first_gap=AP_cont_timestamps(gaps(1)+1);
num_samples_first_gap=tend_first_gap-tstart_first_gap;
startByte_first_gap = 2*384*(int64(gaps(1)));
fid = fopen(contFile, 'r');
fseek(fid, startByte_first_gap, 'bof');
signal_first_gap = fread(fid, [384 num_samples_first_gap], '*int16');

Repeat for 2nd gap.

jsiegle, 2021-07-29:
You can send the xml to [support@open-ephys.org](mailto:support@open-ephys.org).

The code you posted looks fine. Do the non-zero values look random or flat? That's expected as long as they don't look like actual data.

Hankslab-ABG, 2021-07-29:
Sent.

It certainly doesn't look random or flat. See the attached picture of channel 1: zeroed regions bracketing the non-zero region.
[image]

jsiegle, 2021-07-29:
Ok, that's a bit strange, but perhaps not entirely unexpected.

Based on the settings.xml file you sent, I have two recommendations that should prevent this from happening in the future:

The first is to remove the two Bandpass Filters from the signal chain. Filtering 768 channels 2x requires a non-trivial amount of processing power, and taking these out will reduce the probability of CPU spike interfering with recording. The data for the AP and LFP bands are already filtered on the probe, and the LFP Viewer includes an offset subtraction option that will center the signals around 0 for visualization purposes.

The other recommendation is to upgrade to the latest version of the GUI (0.5.4). The "Record Node" architecture introduced in v0.5.0+ was specifically designed to make Neuropixels recordings more efficient, so I think you'll find significant performance improvements after switching. Be sure to read through [this documentation page](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/index.html) to better understand how it works.

Hankslab-ABG, 2021-07-29:
Thanks very much for the suggestions! We're not actively collecting data right now, but this is super helpful as far as optimizing future recordings.

Right now our biggest concern is making sure these drops are detected and that they won't interfere with our interpretation of the data. From what I can tell there shouldn't be any frame-shift errors induced by this with regards to aligning kilosorted data with our behavior rig's TTL sync pulses, correct? It doesn't seem like it should, but any additional perspective is welcome.

With regards to building in quality-control checks, I'm assuming that checking each continuous timestamp file for gaps and comparing the total number of samples to the number of samples in the binary file are a valid way of detecting this drops. Would you agree? To that end, is there a scenario you could imagine that might result in the binary file losing sample elements in the same way as the timestamps.npy file and thus inducing a frameshift?

jsiegle, 2021-07-29:
That's right...because the timestamp indicates the index in the .dat file (after shifting by the first timestamp), the timestamps for the TTL events will be consistent with the .dat file even if some of the data was dropped.

Comparing the number of elements in the continuous timestamp file with the number of samples in the .dat file is the right QC approach. If they match, then everything should be fine. Otherwise, you can expect dropped/invalid data wherever there are gaps in the timestamps. I don't think it's possible for samples to be missing from the .dat file without corresponding gaps in the timestamps—except in the case I mentioned previously where the PXI data buffer overflows. But I don't think that's an issue here, and that's very unlikely to happen when acquiring from a single probe.

Hankslab-ABG, 2021-07-29:
Sorry, just to clarify: When you say " I don't think it's possible for samples to be missing from the .dat file without corresponding gaps in the timestamps", are you referring to the samples missing entirely or just be zeros/flat?

I guess what I'm getting at is if there's any chance the .dat file might be entirely missing elements and thus concatenating non-consecutive indices in such a way that the recording appears shorter than it actually was from kilosort's perspective (which assumes the beginning of the recording is time 0 and each successive sample is incremented by 1, since it doesn't make use of the timestamps.npy file at all).

jsiegle, 2021-07-29:
There are two possible causes for dropped data, errors on *reading* or errors on *writing*. Reading errors would be caused by a buffer overflow on the PXI card, and would manifest as actual missing samples (not zeros), but with consecutive timestamps that are the same size as the continuous data.

Writing errors are caused by the Record Node not saving one or more buffers in a row, and manifest as non-consecutive timestamps and blocks of zeroed out continuous data. It definitely looks like this is what you're seeing, rather than reading errors.

Hankslab-ABG, 2021-07-30:
Okay that's a relief to hear that it's the latter and not the former and that the reading errors are unlikely given our setup.

One question about that though: For the reading errors would there be any way to know that one actually occurred or would this be a silent error that would compromise an analysis without us ever knowing? If I'm understanding you correctly that the there would be missing samples in the binary file, and adjacent samples ended up being non-adjacent in time, but the timestamps.npy file read as if there were no interruptions, the entire timeframe of the data would be thrown off without any way to correct for it or know it happened. Is that correct or am I missing something?

jsiegle, 2021-08-01:
If you are sending some type of sync pulse throughout the recording, and these pulses are unambiguous (either because they have a unique temporal pattern, or have irregular intervals), then it's possible to detect the dropped segments and realign everything. This is what happened in our case when we experienced the Neuropixels buffer overflow. We were able to match the ["barcodes"](https://github.com/open-ephys/sync-barcodes) on the Neuropixels sync line with the barcodes on the master clock (recorded by a separate NIDAQ device), and recover the correct timestamps for every interval except the 30 s block in which the reading error occurred.

So, if you only sending sync pulses at the beginning and the end of the experiment, then reading errors would be undetectable. But if there are multiple sync events that are also recorded by another device (such as a BPod), then the missing samples could be accounted for.

Hankslab-ABG, 2021-08-04:
So is it the case for a read error that the event channels on a given processor would also lose samples? Regardless, is there any reason to believe that a read error would cause the event timestamps to no longer be in alignment with the continuous timestamps? If the continuous and event timestamps (and thus the continuous and event data) would still be in agreement following a read error then it sounds like our data for that session would still be usable because the timestamps from the event data (which we use to sync our trial start times) would still point to the correct portion of the continuous data. Am I thinking about this correctly?

Similarly, if there's a read error does it effect multiple processors or just one? I ask because I'm wondering if the LFP data we record on a separate processor (100.1) would also lose samples or if some reason it might be unaffected due to the lower sampling rate.

Hankslab-ABG, 2021-08-09:
Hi Josh,

So a further update to this:

I've examined the LFP and AP timestamps (those that weren't deleted) as well as the sync_messages.txt file from all of our recording sessions for 2 rats.

I found the following and was wondering if you could provide some context in addition to the questions from my last post:

For the majority of sessions in which the AP timestamps file was not deleted, the difference between the first timestamp in the AP timestamps and the LFP timestamps is 11 samples. However, in some sessions this goes as high as 787 samples, with the mean difference being 31.36 samples. Interesting, those sessions which have an increased disparity in the first AP vs LFP timestamp also have a greater mismatch between the first timestamp of a given band and the sync_messages.txt start time. For instance, in the session in which the first timestamp of AP and LFP bands differed by 787 samples, the first LFP timestamp differed from the LFP-band sync_messages start time by 51 samples and the AP timestamp differed from the AP-band sync_messages start time by 1024 samples. Does this make sense given what you know about the source of the discrepancy between the sync_messages file and the timestamps?

jsiegle, 2021-08-09:
Ok, based on what you've described, it's clear that with the old recording architecture (pre-GUI v0.5.0), it is possible for the LFP and AP band .dat files to start writing at different times, and the discrepancy between the first AP and LFP timestamps is correlated with larger discrepancies between the timestamp in `sync_messages.txt` and the actual first AP/LFP timestamp. We have not seen this behavior in our testing with the more recent Record Node system, but that's obviously not helpful for the data you've already recorded.

I'm assuming your main issue is figuring out how to recover the true AP band timestamps in cases where the AP `timestamps.npy` file has been deleted? In that case, you will likely need to look for correlated signals between the AP band and the LFP band, and use those to align the samples. Fortunately, it is quite common for large spikes to leak into the LFP band, so you should be able to use those to determine the true offset between the start of each .dat file. Can you see if that works for one of your sessions?

Hankslab-ABG, 2021-08-09:
Yes, this is all in the service of recovering the first AP timestamp so that we can align our event data (TTL based trial sync signals) with the kilosorted AP band data. We can give your suggestion a try and let you know how it goes. Can we assume that even if the writing begins at different times for processor 100.0 vs 100.1, the reading should start simultaneously?

Also, did you have any thoughts about my previous post regarding some additional read error and filtering questions?

Thanks!

jsiegle, 2021-08-10:
Yes, the reading of both the AP (100.0) and LFP (100.1) starts at the same time.

With regard to your previous post, the short answer is that the continuous and event timestamps would still be aligned following a read error, and the AP and LFP band would both be missing samples during the same intervals. But I think it would be most helpful to draw a diagram to explain this, and I haven't gotten around to that yet.

Hankslab-ABG, 2021-08-24:
Okay, great. That's good to know.

One additional (hopefully nearly last) question: Within version 0.4.6 can you point us to the part of the code where sync_messages.txt is written? We want to understand more thoroughly how it's created and where the discrepancies arise from.

Hankslab-ABG, 2021-09-20:
Just wanted to follow up on this before the issue is closed: 

@jsiegle Within version 0.4.6 can you point us to the part of the code where sync_messages.txt is written? We want to understand more thoroughly how it's created and where the discrepancies arise from.

Thanks,

-Adam

jsiegle, 2021-09-20:
This is where the sync messages get written: https://github.com/open-ephys/plugin-GUI/blob/07391b0f3aa1343ef17aa0ed7efffa5737b24dd7/Source/Processors/RecordNode/BinaryFormat/BinaryRecording.cpp#L555-L561

And this is where they are generated by each source processor: https://github.com/open-ephys/plugin-GUI/blob/07391b0f3aa1343ef17aa0ed7efffa5737b24dd7/Source/Processors/GenericProcessor/GenericProcessor.cpp#L699-L707

jsiegle, 2022-07-30:
We have implemented a number of changes in GUI version 0.6.0 to guarantee that all sync messages and timestamps files are opened at exactly the same time. If you're still experiencing this problem with the latest version of the GUI, please open a new issue.

closed 2022-07-30 by jsiegle