open-ephys/plugin-GUI #467: Start Time Precision

afimxat, 2021-08-10:
Is it possible to get the start time of the recording in milliseconds precision? 
GUI version: 5.4

 

start time in settings.xml file:
```
<DATE>22 Jul 2021 11:07:26</DATE>
```
sync info in sync_messages.txt of recording1 folder:
```
Software time: 162908@10000000Hz
Processor: Rhythm FPGA Id: 100 subProcessor: 0 start time: 163072@30000Hz
```

jsiegle, 2021-08-11:
Hi! In the current version of the GUI, second-level precision is all that's available. The "software time" is based on a 10 MHz clock that periodically loops back to zero, so it can't be used to determine the absolute start time.

In the next release (0.6.0), we will add the millisecond-level start time to both the `settings.xml` and `sync_messages.txt` files

afimxat, 2021-08-11:
Ok, good to know. Thanks!

closed 2021-09-29 by afimxat