open-ephys/plugin-GUI #468: PXI Trigger input was not recorded

Linoki, 2021-08-11:
Hi I am using the version 0.3.3-API7 for neuropix-PXI as plugin. When I use an external TTL as a synchronization signal for neuropixel and National instrument, I dont get the TTL recorded in the 385th LFP_SYNC channel. Connecting to the sma port only brings the TTL signal amplitude down in the national instrument.

Anyone knows how to get the TTL registered in neuropixel LFP?

[image]
[image]

medengineer, 2021-08-16:
Hello,

It appears the Neuropixels-PXI plugin is not detecting the synchronization signal at all. Is the external TTL signal at least 5V? Are you able to see the TTL signal in the Neuropixels' LFPViewer without the National Instrument device connected? The sync signal should appear as yellow vertical bars when the signal goes high.

jsiegle, 2022-07-30:
This sounds like a hardware issue. If it's still happening, please open a new issue in the [`neuropixels-pxi`](https://github.com/open-ephys-plugins/neuropixels-pxi) repository.

closed 2022-07-30 by jsiegle