open-ephys/plugin-GUI #469: Rhythm FPGA impedance testing interface displays a maximum of 41 channels per headstage

jsiegle, 2021-08-18:
Example here:

[image]

Milanbecker, 2021-12-16:
It has been a long time but is there any solution for this problem as I am struggling with the same issue at the moment

anjaldoshi, 2021-12-17:
Hi,

We have a fix for this on our development pipeline and is ready for our next release v0.6.0. Since a stable release will take some time, we went ahead and patch-fixed the plugin for you. 

I'm assuming you're on Windows and installed the GUI (v0.5.0 or higher ) with the installer. To install this patched plugin on your end, first you'll need to unzip this [RhythmNode.zip](https://github.com/open-ephys/plugin-GUI/files/7736956/RhythmNode.zip) file, and then copy the `RhtymNode.dll` and replace it with the file of the same name in `C:\Program Files\Open Ephys\plugins`.

Let me know if you any further issues/questions.

- Anjal

Milanbecker, 2021-12-19:
Thank you so much for the quick reply this has worked perfectly!

edmundchong, 2022-03-02:
hi! I was wondering it there was a fix for Mac OSX too? I'm having the same issues. Thanks!

jsiegle, 2022-03-03:
Here's a version of the Mac app with that fix implemented: https://www.dropbox.com/sh/fp6ua03x4uxrm33/AABKjhSWqrrL4IFaRwBOBUQha?dl=0

The next release (0.6.0) will include this fix, as well as a number of other improvements to the Rhythm FPGA module (which will actually be renamed "Acquisition Board" and "Intan RHD USB" depending on which hardware you're connecting to).

edmundchong, 2022-03-04:
@jsiegle Thanks Josh! It works perfectly now

jsiegle, 2022-07-30:
This is now fixed in the officially released version of the GUI.

closed 2022-07-30 by jsiegle