open-ephys/plugin-GUI #470: Windows log file

adredish, 2021-08-19:
There should be a flag/process through which the debug log can be written to a log file on Windows.  I found when building a new plugin, that I had a bug that crashed the OpenEphys signal chain.  Tracking it down was extremely difficult because when the program crashed, all my debug info was lost.  

Moreover, for reproducibility purposes, I would like to store this debug log with each day's data.

thx

anjaldoshi, 2021-08-19:
Hi David,

We recently introduced a thread-safe logger in the GUI in v0.5.3 and we are planning to use it for generating and outputting debug log to a file during every run of the GUI in a future release. This will definitely help in debugging the GUI as well as the plugins that are loaded into the GUI. Since debug logs can help only up to a certain limit, we would still advice using a proper debugger to debug the GUI or its plugins.

How are you debugging your plugin right now? On Windows, we use the Visual Studio's debugger to track down any bugs or crashes in the GUI / plugins. For debugging an external plugin, we build both the GUI and the plugin in debug mode, and then run the GUI with the debugger. This is really helpful when you are encountering random crashes as the debugger will break and show the entire call stack as well as the exact line on which the GUI crashed.

adredish, 2021-08-19:
I actually solved the plugin problem by stepping back and rebuilding from
an earlier plugin that I found in that worked.  I actually have it working
now.  So this isn't a crisis for me, but I felt it was something that we
should have working for reproducibility and future cases.
Thanks for the very quick reply.
adr

anjaldoshi, 2021-08-19:
Good to know that you were able to resolve your issue and thanks for giving your input on this. I'll keep this issue open until we have a proper logging system in place.

jsiegle, 2022-07-30:
The latest version of the GUI outputs a log file that includes every action the user has taken, as well as debug output written using the `LOGD` command.

closed 2022-07-30 by jsiegle