open-ephys/plugin-GUI #471: Feature request: user interface for manual cable delay adjustment

cosmouse, 2021-08-23:
Hello,

We have had occasional issues with certain tethers not working correctly due to apparent initialization errors (large amplitude "noise" on some or all channels). Bonsai has a 'CableDelay' setting that can be used to manually set the MISO cable delay value, and that seems to fix our issues. Would it be possible to similarly have user-level access to the cable delay values within the Open Ephys GUI?

Thanks!
Phil

jsiegle, 2021-08-23:
Yep, this is a good idea! We're planning some updates to the Rhythm FPGA module for the next release (0.6.0). We'll make sure this gets included!

cosmouse, 2021-08-24:
Awesome, thanks Josh!

jsiegle, 2022-07-30:
Hi @cosmouse, we haven't had a chance to implement this yet, so I opened a new issue in the Rhythm Plugins repository: https://github.com/open-ephys-plugins/rhythm-plugins/issues/2

closed 2022-07-30 by jsiegle