open-ephys/plugin-GUI #479: Signals of all channels are the same.

YadongD, 2021-09-22:
Hi,

I am new to ephys study and the Open-Ephys system. I'm using GUI v0.5.5. Recently, I plugged in two animals, both are with 64 channels. 
My problem right now is when starting acquisition, the data of all channels from the LFP viewer looks the same(I would say these same signals have spike patterns). I don't know if there are any specific settings in the GUI that might help solve this or it might be some other reasons.

Thank you!

Best,
Yadong

jsiegle, 2021-09-22:
Hi Yadong, can you post a screenshot of the LFP Viewer, and send your signal chain XML file (lastConfig.xml or settings.xml) to support@open-ephys.org?

YadongD, 2021-09-22:
> Hi Yadong, can you post a screenshot of the LFP Viewer, and send your signal chain XML file (lastConfig.xml or settings.xml) to [support@open-ephys.org](mailto:support@open-ephys.org)?

Thank you so much, Josh! 
Email sent. Let me know what else should I provide.

closed 2022-07-30 by jsiegle