open-ephys/plugin-GUI #482: Anyway to recover corrupted data?

priyanka-cshl, 2021-10-15:
I am analyzing a batch of data I acquired over the past few months and noticed that some of my recording sessions are corrupt - despite using the same signal chain, GUI settings etc. Corruption happens rarely - on random days, and on random sessions within the same day.
I noticed there are several issues related to this topic and solutions that potentially prevent this from happening but did not find an answer as to whether there's any way to recover the corrupted data files.
Some of these sessions are precious - Is there any way I can read the data, even if some chuncks are lost etc?

Thanks!

jsiegle, 2021-10-15:
Hi Priyanka – it depends on which format you're using, and what happened to cause the corruption. Can you share the data for one example session with corrupted files?

Claybarn, 2021-10-15:
In case this is helpful, I use this function on my npy files when I run out of disk space (oops):
https://github.com/Claybarn/salvage_npy

If I remember, I think the .dat files are still readable, but totally depends on what kind of corruption you are talking about.

jsiegle, 2021-10-15:
I'm able to load these files by ignoring the standard checks for file integrity (line 125 and 146 in [OpenEphys.py](https://github.com/open-ephys/analysis-tools/blob/master/Python3/OpenEphys.py)). They have about 260 seconds of valid data, and the remaining samples are all zeros.

Was there anything odd about these sessions, or did they seem normal until you went to look at the data?

priyanka-cshl, 2021-10-15:
That’s unfortunate. They should have approximately an hour long stretch of data.
This means the data got corrupted before saving.
Nothing odd about this session, everything looked normal while recording.
Didn’t realize until today when I actually tried to load the files.

priyanka-cshl, 2021-10-16:
Will definitely check. 
Thanks for the help!

jsiegle, 2022-07-30:
We've now implemented a number of changes that make data corruption like this much less likely to occur. If you see any evidence of corrupted data when using the latest version of the GUI, please open a new issue.

closed 2022-07-30 by jsiegle