open-ephys/plugin-GUI #483: Can't see Spike Sorter detected waveforms in Spike Viewer

ahleighton, 2021-11-03:
Hi, 

Using version 0.5.5.2 or 0.5.5.1 on Windows, if I have a Spike Sorter followed by a Spike Viewer in my signal chain, I can't see the waveforms in the Spike Viewer, even if I copy the [Signal Chain Tutorial](https://open-ephys.github.io/gui-docs/User-Manual/Building-a-signal-chain.html#buildingasignalchain). I can still see the PCA, and the number of detected spikes there varies with the threshold I've set in the Spike Sorter. 

[image]

anjaldoshi, 2021-11-03:
Hi,

Thanks for bringing this to our notice. We tried reproducing this bug, and we can confirm it does in fact occur when using Tetrodes in the Spike Sorter. We have figured out a fix for this, and we'll put out a patch release for it today. 

- Anjal

anjaldoshi, 2021-11-05:
The patch release is out and can be found here: [v0.5.5.3](https://github.com/open-ephys/plugin-GUI/releases/tag/v0.5.5.3)

closed 2021-12-01 by anjaldoshi