open-ephys/plugin-GUI #484: Open Ephys GUI crashes at launch

bennycui99, 2021-11-10:
My GUI still crashes at launch at the 0.5.5.3 version.
Here is the information on the console I copied

Debug console...

Created processor graph.

Got audio device.
Audio device name: Speakers (2- High Definition Audio Device)
Audio device type: Windows Audio
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024
Created audio component.
Created message center.
Created info label.
Created graph viewer.
Created data viewport.
Created editor viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: ArduinoOutput...
Loading Plugin: BasicSpikeDisplay...
Loading Plugin: CAR...
Loading Plugin: ChannelMappingNode...
Loading Plugin: EvntTrigAvg...
Loading Plugin: FilterNode...
Loading Plugin: IntanRecordingController...
Loading Plugin: LfpDisplayNode...
Loading Plugin: PhaseDetector...
Loading Plugin: PulsePalOutput...
Loading Plugin: RecordControl...
Loading Plugin: Rectifier...
Loading Plugin: RhythmNode...
Loading Plugin: SerialInput...
Loading Plugin: SpikeSorter...
Num built in engines: 2
Adding engine: Binary
Adding engine: Open Ephys
Num plugin engines: 0

Loading window bounds.

Comparing configs
Creating processor with name: Rhythm FPGA
Creating from plugin info...
Rhythm FPGA(1)::Rhythm FPGA
C:\Program Files\Open Ephys
---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----

FrontPanel DLL loaded.  Built: Nov 11 2016  13:08:03

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6310LX45 with serial number 2049000V24

Trying to open device with serial 2049000V24
Device opened
FPGA system clock: 0 MHz
Opal Kelly device firmware version: 1.30
Opal Kelly device serial number: 2049000V24
Opal Kelly device ID string: Opal Kelly XEM6310

Rhythm configuration file successfully loaded.  Rhythm version number: 1

Initializing acquisition board.
Blocksize set to 1024
DDR burst set to 32
Sample rate set to 30000
USB3 board mode enabled
Sample rate set to 30000
Number of enabled data streams: 8
Checking for connected amplifier chips...
Number of enabled data streams: 0
Sample rate set to 30000
Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 0
Creating processor with name: Record Node
Creating from plugin info...
(-1)::Record Node
Record Node updating settings.
Creating processor with name: LFP Viewer
Creating from plugin info...
LFP viewer(1)::LFP Viewer
LFP Viewer updating settings.
Creating processor with name: File Reader
Creating from plugin info...
(-1)::File Reader
File Reader updating settings.
File Reader setting num outputs to 16
Loading parameters for Rhythm FPGA
Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 8
ADC Button toggled
Loading parameters for File Reader
Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 8
Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 8
Record Node updating settings.

When I launch the GUI without the acquisition board connected, I can open it and it pops out the message"Acquisition board not found". When I connect the acquisition board, it crashed again.

jsiegle, 2021-11-10:
Is your acquisition board being disconnected via USB or the 5V power supply? When I disconnect just the USB, I can replicate these crashes. However, when I disconnect and re-connect the power supply, the signal chain reloads fine.

If you are power cycling the acquisition board between launches and it's still crashing, can you send the `lastConfig.xml` file that's giving you issues?

sppmg, 2021-11-11:
Hi
I got similar problem, but it's ` Intan headstages` which not connected.
Crash immediate after launch, no any warning.

jsiegle, 2021-11-11:
Hi @sppmg, if you remove `lastConfig.xml` and `recoveryConfig.xml` from `C:\ProgramData\Open Ephys\` (be sure to save a backup somewhere), you will be able to open the GUI and re-build your signal chain. If you can send a copy of either one of those files to `support@open-ephys.org` we can get a better sense of what's going on.

sppmg, 2021-11-12:
Hi.
*In my case*, it's fixed after I connect `Intan headstages`. (It doesn't need remove `*.xml`.)
But I think if some warning or maybe `pointer check`/`exception handing` will better.

jsiegle, 2021-11-12:
Definitely agree! We are currently working on cleaning up the Rhythm FPGA plugin, this will be fixed in the next release.

bennycui99, 2021-11-17:
@jsiegle When I removed `lastConfig.xml` and `recoveryConfig.xml` from `C:\ProgramData\Open Ephys\` I still could not launch the GUI. The debug console is different:

Debug console...

Created processor graph.

Got audio device.
Audio device name: Speakers (2- High Definition Audio Device)
Audio device type: Windows Audio
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024
Created audio component.
Created message center.
Created info label.
Created graph viewer.
Created data viewport.
Created editor viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: ArduinoOutput...
Loading Plugin: BasicSpikeDisplay...
Loading Plugin: CAR...
Loading Plugin: ChannelMappingNode...
Loading Plugin: EvntTrigAvg...
Loading Plugin: FilterNode...
Loading Plugin: IntanRecordingController...
Loading Plugin: LfpDisplayNode...
Loading Plugin: PhaseDetector...
Loading Plugin: PulsePalOutput...
Loading Plugin: RecordControl...
Loading Plugin: Rectifier...
Loading Plugin: RhythmNode...
Loading Plugin: SerialInput...
Loading Plugin: SpikeSorter...
Num built in engines: 2
Adding engine: Binary
Adding engine: Open Ephys
Num plugin engines: 0

Loading window bounds.

Num engines: 2

jsiegle, 2022-07-30:
This should be fixed now, but if not please open an issue in the new Rhythm Plugins repository: https://github.com/open-ephys-plugins/rhythm-plugins

closed 2022-07-30 by jsiegle