open-ephys/plugin-GUI #485: v0.5.5.3 will generate 1Hz pulse

sppmg, 2021-11-12:
Hi
I connect signal terminal (MultiClamp 700B output) to ADC_1 and also connect to oscilloscope.
When I play/record on open-ephys, oscilloscope will show some pulse (and maybe it's a pulse train).
*(Note, also happened when ADC_1 only connect to oscilloscope)*
[image]

That reminds me the voltage spikes from inductive. **But this pulse will not generate when I use `v0.4.6` in same connection.**

Oh, and this pulse will fast generate when measuring impedance.
------- 
some spectrogram (10KHz, ~5sec):
open-ephys recording (ADC_1, to oscilloscope):
[image]

open-ephys recording (ADC_8, float/no connection):
[image]

open-ephys recording (ch_7, probe channel, few channel has some effect.):
[image]

jsiegle, 2021-11-15:
Hi @sppmg, thanks for documenting this. Based on the images you've shown, it doesn't seem like there's any obvious contamination in the recorded data. The 1 Hz artifact is clear in the floating ADC channel, but this won't matter for an actual experiment.

Can you check whether the voltage spikes measured with the oscilloscope go away when you disable the LEDs on the acquisition board? There is a button to toggle this within the Rhythm FPGA plugin editor.

@aacuevas, are the any other changes in the bitfiles that could explain the difference between v0.4.6 and v0.5.5.2?

aacuevas, 2021-11-15:
My guess would be cross-signal from the HARP output we added recently. It is sent through the i2c lines, so there could be coupling with some nearby channels. I also checks with this pulse being generated every second.

sppmg, 2021-11-16:
Hi
Yes, I did turn off  LED, same result. No any difference (amplitude or
frequency).

And in my memory, spike also show in oscilloscope, even if it's no in
recording. (Maybe the spike source after ADC module/block? It's just my
guess.) It may affect another device which connect together.

sppmg, 2021-11-17:
I will check again about does spike affect on connect together device.
(It's doesn't see in real *animal* experiment on yesterday. Previous testing experiment use saline.)

jsiegle, 2022-07-30:
It seems like this artifact is benign...if it does seem to be affecting your experimental data, please re-open this issue.

closed 2022-07-30 by jsiegle