open-ephys/plugin-GUI #486: empty sync_messages file

dwhzlh2021, 2021-11-30:
Today when I process the data, I found I have an empty sync_messages file, is this a known bug? and is there any way I can figure out what time I started recording so I do not waste the data? Thanks in advance.

medengineer, 2021-11-30:
Hi,

This is a known bug and will be fixed in the next release. However, the sync timestamps are redundant with the first recorded timestamp for each processor in its respective timestamps.npy file so you can recover the same values from there.

dwhzlh2021, 2021-11-30:
Thank you so much!

closed 2021-11-30 by dwhzlh2021

geffenlab, 2022-03-09:
Hi, I just encountered the same issue with one of our recordings. As a sanity check, I opened an older recording with an intact sync_messages.txt file and compared the values in the timestamps.npy files to the start times.

For the spike processor, the timestamps match:
`Processor: Neuropix-PXI Id: 100 subProcessor: 0 start time: 400225@30000Hz`
and the first value of timestamps.npy is 400225

However, for our events card, the timestamps do not match:
`Processor: NI-DAQmx Id: 102 subProcessor: 0 start time: 251001@30000Hz`
while the first value of timestamps.npy is 249001... off by exactly 1000...

@medengineer Any advice here?