open-ephys/plugin-GUI #487: Cannot install Event Broadcaster, Network Events, ZMQ Interface, or Crossing Detector plugins in Windows

fusio81, 2021-12-01:
With a new installation of GUI on Windows 10, when I attempt to install the Event Broadcaster, Network Events, or ZMQ Interface plugins, the installation fails with the following message on the console:

"D:\a\plugin-GUI\plugin-GUI\Source\Processors\PluginManager\PluginManager.cpp:219: Failed to load plugin DLL: DLL Error 0x7e Loading Plugin Failed!!"

I've tried on two different Windows 10 machines with the same results. Is there a work-around so that I can install these? Also, when I try to install the Crossing Detector plugin, GUI crashes (quits unexpectedly). On the other hand, I'm able to install all of these plugins without any problems on MacOS.

anjaldoshi, 2021-12-01:
Hi,

This is a known issue for plugins with dependency on the ZMQ library. This happens when there are missing runtime libraries on Windows. Can you install the [VS2013 C++ restributable packages](https://www.microsoft.com/en-us/download/details.aspx?id=40784) and see if that resolves the loading error for Event Broadcaster, Network Events, and ZMQ Interface plugins.

Regarding Crossing Detector, I fixed a naming issue on Artifactory causing this crash. You should be able to install it now without any issues.

Let me know if you have any other issues.

Best,
Anjal

superior-colliculus, 2021-12-03:
Thanks-- that solved it! 

I had already installed 2015, 2017, 2019, and 2022 per the instructions on the Wiki, and when I added VS2013 the problem went away. Crossing Detector seems to work for me too now.

Best,
George

stinos, 2022-07-27:
On a Windows 11 machine (could be coincidence) I actually had to install the redist version from [KB 3138367](https://support.microsoft.com/en-us/topic/update-for-visual-c-2013-and-visual-c-redistributable-package-18d278e8-1c79-2630-e8ae-e9f4679b2393), not the one linked above, to get this working. The one above did work for Windows 7 and 10 machines.

anjaldoshi, 2022-07-27:
Thanks for the tip!

Good thing is, this is fixed in v0.6.0 of the GUI. You won't have to worry about installing separate redistributable other than the one that is shipped and installed with the GUI.

closed 2022-07-30 by jsiegle