open-ephys/plugin-GUI #488: Feature Request: Recording at different sampling rates simultaneously (ie. one for spikes, one for LFP)

jclarin, 2021-12-01:
For some experiments, it is necessary to record at two different sampling frequencies: one for spike data and another for LFPs. I've seen others asking about this, so it seems like the community as a whole would benefit from this functionality. While I'm not sure of the technical difficulty of implementing this, I could envision a plugin that would down-sample the data stream in real-time. Then one could simply place record/visualizer nodes before and after the down-sampler. If there is a workaround for accomplishing this, I'd be interested in hearing any suggestions. 

Thanks!

- Jake

jsiegle, 2021-12-02:
Hi Jake, thanks for the suggestion, we always appreciate user feedback! 

Technically this is not too challenging – the `AudioNode` already implements a resampling operation inside its [`process`](https://github.com/open-ephys/plugin-GUI/blob/dbe5892d2cb74f851503f667edbad056f71ebb83/Source/Processors/AudioNode/AudioNode.cpp#L245-L490) method in order to match the sampling rate of the incoming data to the sampling rate of your computer's audio card. It would be straightforward to use the same algorithm inside a downsampling plugin. 

However, it's best practice to save all the data for every channel, and perform any filtering/downsampling offline. You can archive the original data if it's not needed for further analysis, but at least it will be available if you need to go back to it. So unless you have an extreme case (e.g., very high channel count, multi-day recordings), I would recommend placing a Record Node immediately after your data source to record the full-sample-rate data. And even in that case, I think it's usually better to buy more hard drives than to throw away data during acquisition.

jsiegle, 2022-07-30:
Closing this issue, since it's generally not a good idea to downsample data online.

closed 2022-07-30 by jsiegle