open-ephys/plugin-GUI #491: Multiple recording nodes store folders with different time stamps

Mateo-Lopez-Espejo, 2021-12-18:
given a recording session that starts at 2021-12-12_16-33-12 the first recording node will write to a folder with the right time stamp, but some times (under high cpu load?) the second recording node (after some signal processing modules) will write to a folder with a time stamp delayed by 1 second i.e. 2021-12-12_16-33-13.
both recording nodes have default path and in theory inheriting the save path from the recording controller at the top of the GUI.

medengineer, 2021-12-22:
Hello,

Unfortunately this is currently expected behavior in the case you describe. Manually moving the contents of the second directory into the first after recording is the best solution for now. 

The next iteration of the GUI has a custom naming scheme that will mitigate this issue.

jsiegle, 2022-07-30:
This has been fixed in the latest version of the GUI.

closed 2022-07-30 by jsiegle