open-ephys/plugin-GUI #492: Rhythm FPGA 'PARAM' not remapping correctly from 32 to 16 channel mode

samantha6rose, 2022-01-04:
I am a new user to Open Ephys and have been using OpenEphys v0.5.5 on Windows to record from an Intan RHD 16-channel headstage. I was initially using the software audio output function and the LFP viewer to monitor the data streaming. However, I'm interested in viewing the data on an oscilloscope and external audio monitor. 

To do this, I'm using the AUDIO OUT interface in the Rhythm FPGA plugin to select 'PARAM' to the corresponding channel I want to listen to and view data on the external oscilloscope, feeding them from Output 1 to a split BNC. This is working fine. 

The issue I've come across is whenever I try to monitor data from by selecting the corresponding 'PARAM' channel, I don't get any signal from Channels 1-8. Channels 9-16 appear on the external oscilloscope just fine (so I know my hardware setup is okay). I can still see data from Channels 1-8 on the LFP viewer and hear the signal through the software audio monitor for all 16 channels, so it isn't an issue with the data streaming itself. 

I am recording in the 16-channel mode. In an effort to troubleshoot, I toggled to 32-channel mode in the Rhythm FPGA plugin. When I do this, the 16 channels of data that I'm streaming appear on Channels 9-24 in the LFP Viewer and software audio monitoring. I then went through and selected each of 'Param' 1-32 for the AUDIO OUT interface and found that, similar to the LFP viewer and software audio, the data was coming in channels 9-24. This leads me to believe the issue may be that when I toggle back to the 16-channel mode, it seems that the Rhythm FPGA plugin is incorrectly remapping the 'PARAM's, despite remapping the audio channels correctly and appearing as 1-16 on the LFP viewer. 

I've also been using a testboard and tape recording to troubleshoot up to this point, so all the data coming in to the 16 channels is identical; because of this it's also not clear to me if the PARAMs I can access in 16-channel mode are CH9->CH1, CH10->CH2..., or CH9->CH9, CH10->CH10... I can try and figure this out, but I'm hoping the rest of the information provided here is enough to get the issue in the plugin resolved? Or that there's a fix for this I've just missed? (I tried my best to check the Google Group and GitHub page for similar issues but had no luck). 

In the mean time I will continue using the LFP viewer and software audio output, but getting this fixed in the Rhythm FPGA plugin would be AWESOME!

Thanks in advance!!

jsiegle, 2022-01-05:
Hi Samantha – thanks for bringing this up! We haven't done much testing with the 16-channel headstages, so it's great to have this info. We will take a look at the source code and see if there's an obvious fix. If so, we can send you a patched version of the plugin. If it's non-trivial, then we will fix it in the next release (v0.6.0), which will be ready in about a month.

jsiegle, 2022-07-30:
The plugin for interfacing with the acquisition board now lives here: https://github.com/open-ephys-plugins/rhythm-plugins

If you're still experiencing this problem with the latest version of the GUI, please open an issue in that repository.

closed 2022-07-30 by jsiegle