open-ephys/plugin-GUI #494: full_words.npy removed from version 0.6?

KatharineShapcott, 2022-01-19:
Hello,
I noticed that the saving of full_words.npy has been removed from the BinaryFormat in version 0.6. In our lab we use TTLs as eventmarkers and therefore need the states of all TTL channels at each new TTL input time. 

Is there a reason why this feature is being removed? Will there be another way to access the states of all channels not just the most recently changed?

Here's a link to the commit I'm referring to:
https://github.com/open-ephys/plugin-GUI/commit/582a24cc61450a53009ac4760602dc1d41a3e8b5

Thanks for your help!
Katharine

jsiegle, 2022-01-19:
Hi Katherine – Thanks for this feedback! We decided to remove it because the information in `full_words.npy` is redundant with `channel_states.npy`, meaning the TTL words can be fully reconstructed offline using the channel states. We will provide code for this in the [`open-ephys-matlab-tools`](https://github.com/open-ephys/open-ephys-matlab-tools) and [`open-ephys-python-tools`](https://github.com/open-ephys/open-ephys-python-tools) repositories. There was also a lot of redundant information in `full_words.npy`, as there is a word saved for every single event, so you need to do some post-processing anyway to recover the unique words.

Would you prefer to keep `full_words.npy` in the event output? Or would it be ok to compute these in a post-processing step? If it's the former, we can easily add this back before the official release.

KatharineShapcott, 2022-01-20:
Hi Josh,
Thanks so much for your quick reply! Unfortunately it's not redundant if you don't know the state of the channels when the recording starts. So for example even if I send 0s to all channels at the start of my experiment, if someone presses record a little bit too late then I can't reconstruct the event markers because I don't know the starting states of the channels anymore. 

Personally I would prefer to keep full_words.npy for safety! But if the state of all channels at the start of the recording would be stored then there would be true redundancy and that would also work.
Thanks!
Katharine

jsiegle, 2022-01-20:
That's a good point! @oepavel was planning to have the Record Node output the initial event state on all channels to account for this, but that hasn't been implemented yet.

In any case, since there is low overhead for saving `full_words.npy`, I think we will add this back in the official release.

KatharineShapcott, 2022-01-21:
That's great to hear, thanks!

jsiegle, 2022-07-30:
In GUI v0.6.0, the full words are now saved for all TTL event channels.

closed 2022-07-30 by jsiegle