open-ephys/plugin-GUI #495: Concatenation of OpenEphys (.dat) and Neuralynx (.ncs) files

saman-abbaspoor, 2022-01-21:
Hi all,

I am trying to concatenate OpenEphys (.dat) and Neuralynx (.ncs) recordings that were collected in a session for spike sorting. OpenEphys recordings had an input range of 250uv and Neuralynx had an input range of 850uv. The sampling frequency for both recordings is 30kHz. I understand that because of the difference in the input range, we should balance the scale factor before concatenation. But other than that, is there anything else I should know? What's the best way to do it? Has anyone done this before? One thing for example is the bit depth of recordings which is different between the recordings. How will that affect spike sorting if files are concatenated? Our plan is to use Kilosort 1 for spike sorting if that is relevant.

Thank you in advance
SAM

jsiegle, 2022-01-21:
Hi Sam – the critical thing is to make sure the volts-to-bits scaling is identical for the two datasets. I would recommend loading both the Open Ephys and Neuralynx data into Python or Matlab and converting them to microvolts (for the OE .dat file, just multiply the raw values by 0.195). Then, you can concatenate the microvolts value into a large array (of floats). This array should then be rescaled (e.g., by dividing by 0.195), converted to 16-bit integers, and re-saved into .dat format which can be ingested into Kilosort.

Let me know if that makes sense, or if you need more details about any of these steps!

saman-abbaspoor, 2022-01-21:
Hi Josh,

Thank you so much for your prompt response. I will give it a try. Just to make sure, after converting to microvolts, I concatenate and re-scale using OE ADC bit (0.195)? Because Neuralynx ADC bit is ~0.026. Also shouldn't I worry about different input ranges?

jsiegle, 2022-01-21:
Either scale factor is fine, as long as you use the same one for both datasets. For 16-bit signed integers, 0.026 µV/bit gives a range of ±850 µV (32768 * 0.026). That's probably fine if your data is high-pass filtered, but will likely saturate for wideband signals. So using 0.195 will give you a larger input range (±6 mV), but will not capture the full resolution of the Neuralynx ADCs. But that shouldn't make a difference, given the average noise level in extracellular data.

closed 2022-07-30 by jsiegle