open-ephys/plugin-GUI #496: spike viewer plot problem on Mac

ZhaoM22, 2022-02-10:


anjaldoshi, 2022-02-10:
Hi, 

Can you try to reproduce this error with v0.6.0 of the GUI? You can download v0.6.0 of the GUI from [here](https://openephysgui.jfrog.io/artifactory/Test/windows/open-ephys-v0.6.0-rc2-windows-beta.zip).  This version is still in 'beta', but we are doing the final release soon and I want to make sure if it got resolved in this version or not. 

\- Anjal

ZhaoM22, 2022-02-10:
Hi Anjal,
The attached file is the Windows version (.exe). The problem happed on the Mac GUI. Could you please send me a linking of v0.60 of Mac GUI?
Thanks a lot!
Mingrui

anjaldoshi, 2022-02-10:
Sorry for the incorrect file. 
Here's the MacOS installer for v0.6.0: https://openephysgui.jfrog.io/artifactory/Test-Installer/mac/Open_Ephys_GUI_v0.6.0-rc2-beta.dmg
And, if you prefer the ZIP binaries, here it is: https://openephysgui.jfrog.io/artifactory/Test/mac/open-ephys-v0.6.0-rc2-mac-beta.zip

jonessess, 2022-02-26:
Thanks- I was having the same problem with spike viewer in 0.5.5.3, so I tried the file you linked (beta of version 0.6.0), and it seemed to fix the display issue. 

However, the 0.6.0 pre-release version also crashed after only a few seconds of use, so I didn't test it thoroughly.

-Mark

jsiegle, 2022-02-27:
What was the signal chain you were using when the crash occurred?

jonessess, 2022-02-27:
I was using File Reader -> Bandpass Filter -> LFP Viewer -> Spike Detector -> Spike Viewer.

Upon further investigation, I found that the crashes seemed to be occurring soon after I changed the electrode list in Spike Detector under MacOS Big Sur 11.6.2. (I had deleted some stereotrode channels and added single electrode channels.)

On the other hand, when I tried the 0.6.0 pre-release with the same signal chain on a different machine using MacOS Catalina 10.15.7, I did not encounter any problems.

-Mark

jsiegle, 2022-02-27:
Thanks, that's very helpful to know! We've been doing most of our Mac testing on Catalina, but will look more carefully for potential OS-specific bugs.

jsiegle, 2022-07-30:
This rendering bug has been fixed by upgrading to JUCE v6.

closed 2022-07-30 by jsiegle