open-ephys/plugin-GUI #497: timestamps of continuous.dat

AGenews, 2022-02-14:
I am truly sorry for this likely naive question.

I haven't given a lot of thoughts to the timestamps.npy files until now. When I load the timestamps.npy of a 67Ch recording sampled @2'000 Hz saved as binary I'll have a 1D array:
108800
108801
...
108866
108867
108868
...
all incremented by 1 ... and this is what I don't get ... is this at the end just samples starting from the software time? Not actual time in µs? I am confused.
sync_messages.txt tells me:
Software time: 108830@1000000Hz
Processor: Rhythm FPGA Id: 100 subProcessor: 0 start time: 109056@2000Hz

All the best and apologies for the trivial question.
Andreas

medengineer, 2022-02-14:
Hi Andreas,

The timestamps represent the sample number since acquisition was started. To get the time in seconds, simply divide the timestamps by the sampling rate. 

The start time should be the sample number at which recording was started. But in your case, it appears there are some timestamps written prior to the start time recorded by the GUI. Can you confirm which GUI version you used in this recording?

AGenews, 2022-02-15:
Dear medengineer,
thanks for your quick reply. I have to admit that I am using an old version v0.4.4.1. In the recording mentioned above I was using RecordControl and gated the recording by a TTL1 which was also started at 109015 according to the dedicated I/O timestamps.npy file, also earlier than what was written in the sync_messages.txt .... likely a buffer related issue?

But I still haven't quite understood something: What would be the use of the timestamps.npy for continuous.dat except for detecting skipped samples? I was expecting e.g. real 'time'stamps in e.g. CPU time, just because of the naming.
Thanks a lot!
Best,
Andreas

medengineer, 2022-02-15:
The mismatch in apparent start times is expected behavior with GUI version 0.4.x. Agreed that the term timestamps here can seem misleading. Saving every timestamp is only necessary for detecting dropped samples. Otherwise, as long as you have the first timestamp you can always reconstruct the remaining ones. 

GUI version 0.5 introduced a plugin-based Record Node that generates a synchronized_timestamps.npy file containing floating point timestamps in seconds which are synchronized with any devices that share a common synchronization signal (more details [here](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/#synchronizer)).

AGenews, 2022-02-15:
Thanks a lot!

closed 2022-02-15 by AGenews