open-ephys/plugin-GUI #498: GUI crashes when adding channel map after neuropixels PXI

haohao1331, 2022-02-15:
Hi, 

I am doing a real-time BCI experiment using neuropixel probes 1.0, and decided to use the falcon output plugin to stream the raw data out of open ephys. However, only the AP band is of interest to us, so I want to remove the LFP band using the channel map plugin. However, putting the channel map plugin after the neuropixels PXI plugin crashes the GUI. 

I'm using open ephys version 0.5.5.3 on windows 10, no base station is connected yet so the PXI plugin is running in simulation mode. To reproduce the crash, start an empty signal chain with the PXI plugin in simulation mode, then drag the channel map plugin after the PXI plugin, and the GUI crashes. I've repeatedly tried this several times and found that about 10% of the time the channel map will be successfully added without crashing, but closing the GUI and reopening will crash again. The windows console closes immediately after the GUI crash, but the last line says something like "channel map update settings". 

Is there a quick workaround for this problem? Or is there another way to remove the LFP channel without using the channel map? 

Thanks! 

Leo

jsiegle, 2022-02-15:
Hi Leo – you should use the Stream Muxer plugin for this instead. The Channel Map plugin is not intended for use with Neuropixels data.

Let us know if you run into any issues with the Falcon Output – we just released that a few weeks ago!

haohao1331, 2022-02-15:
Thanks! I'll try the stream muxer, and so far the falcon output is working great. 

Also, is there a way to down-sample the data? We want to record raw data at 30-40kHz for offline analysis, but we only require about 10kHz for streaming.

jsiegle, 2022-02-15:
No, there's not a downsampling module currently. I also would advise against doing this, as downsampling (even to 20 kHz) will lead to many missed spikes.

In the next version of the GUI (0.6.0), it will be possible to set up a single electrode for every channel of the probe and stream the spike times via the EventBroadcaster plugin. That will greatly reduce your data throughput, while still ensuring that you can capture all of the threshold crossings.

haohao1331, 2022-02-21:
So I tried using the stream muxer and it is working properly with falcon output. However, when there is more than one probe connected, all AP channels will be in the same stream group in the stream muxer, and I can only select one of the probes (384 AP channels) to pass downstream at a time, which is kind of a problem for downstream analysis since we want to access data from all probes together. 

We are considering to use 4 neuropixel probes for recording and streaming in the future, so is there a way to batch all AP channels from each of the probes (4*384=1536 AP channels) and let falcon output stream these together? In order to do this with the current stream muxer seems like I need to split from the neuropixel PXI into 4 streams, use 4 stream muxer to select all the AP channels, and merge these 4 streams into falcon output. Is there a better way to do this? 

Also, we tested with a broken dummy neuropixel probe (the shank snaped, but still can get recording in the GUI). Once the GUI is opened and the probe is detected, clicking the "open settings" buttons of the neuropixel PXI plugin makes the GUI stop responding. Do you have any idea how to fix this? 

Thanks!

jsiegle, 2022-02-22:
That's right, the Stream Muxer approach won't work for more than one probe, unfortunately.

When using multiple probes, the LFP channels will account for such a small fraction of the overall data rate, you could try including them in the stream and just ignoring them on the other end. Also, a future update to the Falcon Output plugin will allow you to select subsets of channels to stream, but we don't have an exact timeline for when that will be ready.

We test the Neuropixels plugin with broken probes all the time, so I'm not sure why yours isn't working. Can you send the output of the debug console prior prior to opening the settings interface?

haohao1331, 2022-02-23:
We are streaming to a localhost so aren't too concerned with data rate at this point, falcon output is able to stream 4 simulated probe with both AP and LFP channels. But the problem that the LFP channel introduces when streaming is that it sometimes skrews up data parsing on the client side. Once the data is received on our python client, it is reshaped into a matrix using the channel number and sample number. The LFP channels don't have as many samples due to their low sampling frequency, so there are a bunch of trailing zeros for those channels. However, for some messages (I'd estimate about 50% of the time), there are 2~3 zeros after each AP channel in addition to the sample number provided, which shifts the next channel in the matrix and makes the data difficult to parse. I'm guessing the reason for this is sometimes the sample number might be different in some channels, and since falcon output takes the first channel's sample number as global, additional zeros are introduced when resizing the audio buffer (https://github.com/open-ephys-plugins/falcon-output/blob/main/Source/StreamPlugin.cpp#L91-L94). 

I got around this problem by modifying falcon output, looking at sampling rate and filtering out LFP channels in the process() function, and it fixed the data parsing problem. Looking forward to the update so falcon output can select which channels to stream! 

Another might-be-related problem also occurs when streaming with certain bin sizes. On a mac, I used falcon output to stream from file reader the sine wave provided in open ephys' resource folder, and used the same way to parse the data on a python client. In the audio settings, with a sample rate of 48000Hz with buffer size of 1024 samples (21.3 ms), each message consists of 853 samples, and the resulting data matrix have shifted channels and it doesn't match with data from the record node. With a buffer size of 960 samples (20.0ms), each message has 800 samples, and the problem doesn't occur. 

So the above are the problems I've encountered when streaming with the LFP channels, they aren't fatal as I have a temporary workaround, but do let me know if I made mistakes in parsing or if there are better solutions. 

As of the neuropixels plugin, I would need a couple of days to get my hands on the probes again to reproduce, will follow up then. Thanks!