open-ephys/plugin-GUI #501: sample rate options for record node

neurochen, 2022-03-09:
Thanks for the amazing work on open ephys. We are using it to record signals from chronic arrays for about 10 hours a day. To have spikes processed accurately, we would like to use 30 kHz sampling frequency in signal sources. As the continuous signals, in our case low-pass filtered LFP, are recorded at the same sample rate, it ends up with an extremely large data size >100G. This poses difficulties for storage and also offline analysis. We are wondering if it's possible to downsample the LFP in the record nodes, say 2 kHz. Thanks.

jsiegle, 2022-03-11:
Hi! There is not currently the ability to perform downsampling within the GUI. I'd also definitely encourage you to keep the full-bandwidth data around if you're analyzing spikes, in case you need to re-do your thresholding offline.

If you are having issues with data storage, you can look into compression libraries such as [`mtscomp`](https://github.com/int-brain-lab/mtscomp) from the International Brain Laboratory and the [MED Format](https://medformat.org/) supported by Dark Horse Neuro.

neurochen, 2022-03-11:
Thanks. This is very helpful.

closed 2022-03-11 by neurochen