open-ephys/plugin-GUI #505: Plugin load error

JulioEI, 2022-04-20:
I am developing a plugin for OpenEphys. If I build the plugin without a custom Editor it works fine. However, if I include a custom Editor (simply a blank one, just changing the width), the plugin does compile properly but when I run open-ephys it returns the following error: 

/media/julio/DATOS/CNN_plugin/plugin-GUI/Source/Processors/PluginManager/PluginManager.cpp:219: Failed to load plugin DLL: /media/julio/DATOS/CNN_plugin/plugin-GUI/Build/Debug/plugins/Ch_interp_OEplugin.so: undefined symbol: _ZN13ChInterpSpace8ChInterp12createEditorEv
 DLL Load FAILED

The plugin name is "ChInterp". 

The Editor.h code is the following:
```
#ifndef __CHINTERPEDITOR_H_DEFINED__
#define __CHINTERPEDITOR_H_DEFINED__

#include <EditorHeaders.h>
//#include "ChInterp.h"

class ChInterpEditor : public GenericEditor, public Label::Listener
{
public:
   ChInterpEditor(GenericProcessor* parentNode, bool useDefaultParameterEditors);
   ~ChInterpEditor();
private:
   ChInterpSpace::ChInterp * processor;
   JUCE_DECLARE_NON_COPYABLE_WITH_LEAK_DETECTOR(ChInterpEditor);
};
#endif 
```

The Editor.cpp code is:

```
#include "ChInterpEditor.h"
#include "ChInterp.h"

ChInterpEditor::ChInterpEditor(GenericProcessor* parentNode, bool useDefaultParameterEditors=true)
    : GenericEditor(parentNode, useDefaultParameterEditors)
{
   setDesiredWidth(250);
}

ChInterpEditor::~ChInterpEditor(){}
```

Thank you in advance!

anjaldoshi, 2022-04-20:
Hi,

Looks like you haven't implemented the `createEditor()` method in your plugin class, `ChInterp`. 

You need to declare the virtual `createEditor()` method like [this](https://github.com/open-ephys/plugin-GUI/blob/master/Plugins/PhaseDetector/PhaseDetector.h#L45) in your plugin class declaration (inside ChInterp.h), and override the method definition like [this](https://github.com/open-ephys/plugin-GUI/blob/master/Plugins/PhaseDetector/PhaseDetector.cpp#L46-L53) in your plugin cpp file.

This should fix the plugin load error for you. Let me know if you need any further help.

JulioEI, 2022-04-20:
Hi Anjal,

thank you for your prompt answer. I did have the `createEditor()` method declared but I forgot to define it in my cpp file. Now it works! Thank you!

Not sure who should I indicate it to, but there is a typo in the open-ephys plug-in tutorial at https://open-ephys.github.io/gui-docs/Tutorials/How-To-Make-Your-Own-Plugin.html#creating-editor-class-files. Right at the end of the "creating-editor-class-files" section, it says to define the `createEditor()` method on the .h file instead of the .cpp file.

Again, thank you for your help!

closed 2022-04-20 by JulioEI

anjaldoshi, 2022-04-20:
Great to hear that! 

Also, thanks for pointing out the typo, I'll fix it.