open-ephys/plugin-GUI #508: OE GUI (v0.5.5.4) crashes when record is pressed

SumedhNagrale, 2022-04-27:
### GUI crashes with record node
Today, We updated the open ephys code to v0.5.5.4 for our lab repository at https://github.com/tne-lab/plugin-GUI/tree/low-latency
While doing the unit testing for the GUI, we found the OE GUI crashes whenever we press record with a record node in the signal chain as show below. 
[image]
This is also true when we simply press play with record node in the chain. 

### GUI works without record node
Further, we observed GUI works (I mean play) without Record Node in the signal chain as shown below
[image]

### Same issue with the code from https://github.com/open-ephys/plugin-GUI (master branch) 
To check _if this issue is caused by additional plugins we installed_ **or** _if it is caused due to the update we performed_. 
We downloaded the Open Ephys GUI code from https://github.com/open-ephys/plugin-GUI (master branch) and build the project in release mode using VS 2017 after the cmake -G "Visual Studio 15 2017 Win64" ..  
We found the same issue in the GUI (crashes with Record node and works without Record node), for reference here is the image
[image]

Further, we repeated this process with VS 2019 and found the issue to be present in the GUI. 

### Debug Console 
If this is useful here is the debug console when GUI crashes
[image]

Please, let me know if you need additional information about it and also, if I did something incorrectly above.

Thank you!
Sumedh

anjaldoshi, 2022-04-27:
Hi Sumedh,

Thanks for reporting this. I was able to reproduce this and I've figure out a fix for it. We missed a message center event channel creation call while backporting a fix from our upcoming release (v0.6.0) which lead to this behavior. The fix is up on the `development` branch in this commit : 73805cb. It'll get merged to master today after we make sure everything looks good.

anjaldoshi, 2022-04-27:
Fix has been merged to master. Updated v0.5.5.4 binaries are available to download now. 
Let us know if you encounter any further issues.

SumedhNagrale, 2022-04-28:
Hi Anjal,

I am really looking forward to the v0.6.0 (especially with the JUCE update). 
Thank you for the fix!

closed 2022-04-28 by SumedhNagrale

anjaldoshi, 2022-05-02:
Hi Sumedh,

We encountered another issue with v0.5.5.4 related to audio monitoring causing crashes and another one related to merger connections. I've pushed bug-fixes for both of them on the master branch and also updated the v0.5.5.4 binaries. Make sure you are on the latest code to avoid any issues related to this.

Regarding v0.6.0, we just did the final beta release of v0.6.0, and the next release of v0.6.0 is going to be the stable release. So, hopefully very soon :)

SumedhNagrale, 2022-05-03:
Hi Anjal, 

Thank you for letting me know about this, I will update it on our repository. 
For v0.6.0, that's a great news! Just spoke to the lab about this, We as a whole are really looking forward to use it.