open-ephys/plugin-GUI #509: Very High ( 50KHz-4MHz) Inconsistent Impedance in EEG

spygaurad, 2022-05-31:
I am getting very high impedance in my EEG system. 
The components of my EEG setup include: EEG acquisition board (open ephys), amplifier/ADC (open-ephys) and easy cap electrodes and breakout board.
I have used alcohol to clean and conducting paste on electrodes to get better result. The impedance is never going below 10kHz as required. 

I have connected pin 22, reference and ground to collect this impedance data.

[image]

jsiegle, 2022-05-31:
Hi @spygaurad – I don't have much experience with human EEG, but I can try to help. 

First of all, are the electrodes connected to ground via a conductive solution or someone's scalp? If they are just sitting in air then very high impedance values are expected.

spygaurad, 2022-06-01:
Thanks @jsiegle for your response. For collecting this data, i have connected electrode 22(channel 22), the ground electrode and the reference electrode on scalp along with conductive solution. These electrodes are connected to EEG breakout board's port 22, GND and REF respectively. The remaining electrodes are on air.

jsiegle, 2022-07-30:
Closing for now; if you're still experiencing this problem, please re-open this issue.

closed 2022-07-30 by jsiegle