open-ephys/plugin-GUI #512: Open Ephys GUI fails to append datestring to the folder name if prepend text is >23 characters.

cnuahs, 2022-06-07:
Recently updated several data acquisition computers from v0.4.4.1 to v0.5.5.3. The behaviour of Open Ephys GUI v0.5.x+ seems to changed (c.f. v0.4.x) whereby the date/time folder name is *not* honoured if the supplied prepend text is >23 characters.

For example,
```
[datapth]/prepend-text_yyyy-mm-dd_HH-MM-SS/
```
now becomes just
```
[datapth]/prepend-text/
```
if the length of `prepend-text` is >23 characters.

This threshold seems to be applied in `RecordNode::generateDirectoryName()` [here](https://github.com/open-ephys/plugin-GUI/blob/17c2f0085445e2dc6ad20cd8c0e37831caf1ebb3/Source/Processors/RecordNode/RecordNode.cpp#L206):

```c++
if (filename.length() < 24) // this is a hack
	filename += datestring;
else
	filename = filename.substring(0, filename.length() - 1);
```

Possibly introduced in commit 9b16050.

Is this a bug? Or is this planned/expected behaviour, but yet to be documented?

Open Ephys GUI version: 0.5.5.3
Operating system: Windows

medengineer, 2022-06-07:
Hi,

This is expected behavior in 0.5.x versions and was originally done to allow custom names to override the date string. It will be fixed in the 0.6.0 release which includes a new naming scheme editor with more control and flexibility.

jsiegle, 2022-07-30:
v0.6.0 has now been released. See [the documentation](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/index.html#global-recording-options) for more info about how to use custom directory names.

closed 2022-07-30 by jsiegle