open-ephys/plugin-GUI #513: Get informations related to crash

fbens47, 2022-06-10:
System: Windows 10 x64
JUCE: v6.1.2

Hi,

I'm in deep trouble with software related to Ephys Acquisition running with OpalKelly FPGA drivers (RHX and OpenEphys). I can install OpenEphys with installer provided on [OpenEphys website](https://open-ephys.org/gui). At launch the Console opens but the GUI stays white:

```
Debug console...

Created processor graph.

Got audio device.
Audio device name: Speakers (2- Focusrite Usb Audio)
Audio device type: Windows Audio
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024
Created audio component.
Created message center.
Created info label.
Created graph viewer.
Created data viewport.
Created editor viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: ArduinoOutput...
Loading Plugin: BasicSpikeDisplay...
Loading Plugin: CAR...
Loading Plugin: ChannelMappingNode...
Loading Plugin: EvntTrigAvg...
Loading Plugin: FilterNode...
Loading Plugin: IntanRecordingController...
Loading Plugin: LfpDisplayNode...
Loading Plugin: PhaseDetector...
Loading Plugin: PulsePalOutput...
Loading Plugin: RecordControl...
Loading Plugin: Rectifier...
Loading Plugin: RhythmNode...
Loading Plugin: SerialInput...
Loading Plugin: SpikeSorter...
Loading Plugin: KWIKFormat...
Loading Plugin: MatlabInterface...
Loading Plugin: neuropixels-pxi...
Loading Plugin: nidaq-plugin...
Loading Plugin: NWBFormat...
Loading Plugin: probe-viewer...
Num built in engines: 2
Adding engine: Binary
Adding engine: Open Ephys
Num plugin engines: 2
Adding engine: Kwik
Adding engine: NWB

Loading window bounds.

Num engines: 4
```

With RHX, I can't even run the installer. On the other side, with Neuropixels hardware, I can run SpikeGLX.

Is there a way to get information about the crash? Or maybe built it from source and running OpenEphys in debug mode?

Yours,

Flavien

anjaldoshi, 2022-06-10:
Hi,

Can you specify which version of the GUI are you using? The JUCE version you specified (v6.1.2) comes with GUI version 0.6.0 which is still in beta. But, the console log looks like it is from GUI v0.5.x. It would make it easier for us to debug if we know the GUI version.

Also, when you run the GUI, does the GUI stay white the entire time and you have to manually close the GUI or does it crash and close automatically?

jsiegle, 2022-07-30:
Hi @fbens47 – if you're still experiencing this problem with the latest version of the GUI (0.6.0), please re-open this issue so we can help you troubleshoot further.

closed 2022-07-30 by jsiegle