open-ephys/plugin-GUI #515: Neuropixels-pxi plugin installation _ GUI 0.6 crashes while installing this plugin.

Aryo-Zare, 2022-07-18:
Hi Everyone

I downloaded & installed the newest GUI version (0.6) from the exe files ( https://openephysgui.jfrog.io/artifactory/Release/windows/ )
The reason I did not use the recommended installation page is that this new version is not yet linked there (https://open-ephys.org/gui ) !

The problem I have is that the program crashes whenever I want to install the Neuropixels-pxi plugin. 
I attached the error report file. The recoveryConfig.xml file could not be attached here, I'll send it via email. After initial launch of the software (without any probe being attached to the hardware), I tried to select the 'neuropixels' as default configuration (instead of the file reader) which caused the program to crash. Hence the error report may also contain this problem as well. Then I launched the plugin installer in file-reader mode, & after trying to install the latest version of Neuropixels-pxi plugin, the whole GUI crashed. 

I had already gathered my data with the older version of the GUI & Neuropixels-pxi plugin. The reason I want the newer version ff this plugin is to get the probe layout (x-y locations ... , output in json format) according to my conventional electrode contact points selection. Then using it for spike sorting algorithms.
[activity.log](https://github.com/open-ephys/plugin-GUI/files/9131496/activity.log)

anjaldoshi, 2022-07-18:
Hi Aryo,

Version 0.6.0 is still in beta and in final stages of testing. A stable version hasn't been released yet. That's why the website https://open-ephys.org/gui isn't linked to GUI v0.6.0 binaries and the GitHub repo latest release (https://github.com/open-ephys/plugin-GUI/releases/latest) is still v0.5.5.4

Regarding the Neuropixels-PXI plugin, it's not clear from the log file why the GUI is crashing when installing the plugin. Can you remove the `plugins-api8` and `shared-api8` directories from `C:\ProgramData\Open Ephys` and then can you try installing the plugin with an empty signal chain?

Aryo-Zare, 2022-07-19:
Thanks Anjal for your feedback.

I removed the 2 directories as you mentioned and reran the plugin installer. I ran this on 'file reader' mode after clearing the signal chain.
I hope that plugin-installer would not make any difference running in 'file reader' or 'neuropixels' modes. As mentioned above running in 'neuropixels' mode needs a probe to be attached, otherwise the whole GUI crashes, and since I don't want to risk handling an expensive probe I ran in 'file reader' mode. But if the other way is needed let me know.
I attached the report files.  github does not support .xml file attachment, hence I converted it to pdf.

[activity_2022-07-19_09-08-34.log](https://github.com/open-ephys/plugin-GUI/files/9138655/activity_2022-07-19_09-08-34.log)
[recoveryConfig.pdf](https://github.com/open-ephys/plugin-GUI/files/9138657/recoveryConfig.pdf)

jsiegle, 2022-07-20:
Hi Aryo -- a couple questions:

- are you able to install any plugins via the Plugin Installer, or does it always crash upon installation? 
- is there a `neuropixels-pxi.zip` file downloaded to `C:\ProgramData\Open Ephys`, or is that missing?

Aryo-Zare, 2022-07-21:
Hi Josh
I just installed the latest version of GUI (0.6) in another computer (a laptop with no hardware attachment) & I ran it in simulation mode. I could install the neuropixels plugin & got the json file I needed. But I would like to continue detecting the reason it crashed in our acquisition setup. 

I checked updating other plugins, all cause crash. Hence it's not specific to this plugin. 
The zip file is not saved to the directory you mentioned but I checked in the laptop with the installed plugin : there is a file `neuropixels_pxi.dll` saved in `C:\ProgramData\Open Ephys\plugins-api8`, but such a file is not in our acquisition setup computer.

I'm suspecting there may a network error accessing the online repository. Is there such an error report in the log files ?    
I just checked I can not access jfrog.io via browser in our computer. Are those plugins saved in this online repository.
In any case I'm communicating with the IT in our institute to give me access to this site. I'll further inform you after I test it following getting this network permissions.

jsiegle, 2022-07-21:
Thanks for this info! We will add an extra check to prevent crashes in cases where jfrog.io cannot be reached.

For now, you can just copy the contents of `plugins-api` and `shared-api` from your laptop to the acquisition computer in order to install any plugins you need.

jsiegle, 2022-07-22:
Definitely agree! We actually already implemented this change: https://github.com/open-ephys/plugin-GUI/commit/d678090cfa5cfef784cfe8ce18ff2539ad7b9885

Aryo-Zare, 2022-07-23:
Great. I would specifically write about the repository website in the error message (for example:  `Access denied to the repo website  http://jfrog.io/  Check your permissions for this website. `).
Since our IT colleagues initially could not figure this out. And a message as 'check your internet connection' would not necessarily guide the user toward a specific website's firewall blockade.
They may think a general internet connection problem is the reason. They should know the repository website and the network admin would specifically give access to that website.

anjaldoshi, 2025-06-09:
GUI v1.0's Plugin Installer now includes additional checks for network connection issues during plugin downloads. If a problem occurs, it displays an alert detailing the network error along with the corresponding HTTP status code to clarify the cause.

Please reopen this issue if you're encountering crashes when downloading plugins with GUI v1.0.

closed 2025-06-09 by anjaldoshi