open-ephys/plugin-GUI #516: I used CMake for generating VS Project Files However, I got 2 solutions file instead of one.

oliver9-9, 2022-07-29:
I used the following CMake CLI command inside the plugin-GUI top-level directory (Windows 10, x64):

> cd Build
> cmake -G "Visual Studio 17 2022" -A x64 ..

However, I got 2 solution files instead of one. These

anjaldoshi, 2022-07-29:
Hi,

Not sure why `open-ephys-GUI-plugins.sln` is getting generated on your end. Can you try removing everything from the `plugin-GUI\Build` directory and run the CMake command again? Does it still generate the `open-ephys-GUI-plugins.sln` file after doing so?

-Anjal

oliver9-9, 2022-07-31:
Thanks for your response. I remove everything, cloned the project again and run the cmake program again. still I have 2 solution files. The plugins one is in the plugins folder. 

open-ephys-GUI.sln 
is in 
plugin-GUI\Build

whereas, 

open-ephys-GUI-plugins.sln
is in
plugin-GUI\Build\Plugins

Folder

jsiegle, 2022-07-31:
Ok, this is actually expected! The `open-ephys-plugins.sln` allows the default plugins to be compiled using the `ALL_BUILD` project inside of Visual Studio. But you should open the `open-ephys-GUI.sln` file, as that will allow you to build the core application as well as the plugins.

closed 2022-07-31 by jsiegle