open-ephys/plugin-GUI #517: v0.6.0 GUI freezes when starting a visual stimulation paradigm using psychopy

Milanbecker, 2022-08-09:
When starting a visual stimulation paradigm on a second screen using psychopy, the acquisition freezes (or at least the display of it). The Gui is still responding but no new data is displayed even after stopping the visual stimulations. Everything works fine in v0.5.5.4.
I'm running the Gui on windows 10, Acquisition of 60 channels at 20kHz.
I switched back to the old version but any help on this would be great!

jsiegle, 2022-08-09:
This one might be tricky to fix, as it's likely related to the way the updated version of Juce is handling the rendering commands. Does the display stop updating even if you hide and re-show the LFP Viewer tab? Does all animation within the GUI stop (including the acquisition clock)?

If at all possible, I'd recommend running Psychopy on a separate machine, as there is a high likelihood of interference between the two processes (e.g. higher processing latency in Open Ephys, dropped frames in Psychopy).

Milanbecker, 2022-08-10:
Thanks a lot for your quick reply! Yes all animations stopped, hiding and re-opening the LFP viewer did not help. Running Psychopy on a different machine is a good suggestion, I am considering doing this in the future.

jsiegle, 2025-06-08:
Hopefully this is fixed with the latest updates to the Juce rendering engine. If you're able to replicate with v1.0, please open a new issue.

closed 2025-06-08 by jsiegle