open-ephys/plugin-GUI #519: Bug for Spike Detector plugin: Cannot change threshold mode for electrode larger  than 19

yaknerv, 2022-08-16:
I'm working with the open-ephys acquisition board these days. Two 64 channel headstages were connected to the board. I added 128 single electrodes in the Spike Detector plugin. If I select all the electrodes, the threshold mode can be adjusted normally (i.e. using uV, STD or MED). But when I tried to adjust the threshold mode for individual electrode, I found it didn't work for electrode channel larger than 19.

A related question is, whether the ADC channels can be processed as electrode channels in the Spike Detector plugin? I tried to do this because I connected a visual stimuli onset signal to one of the ADCs. I want to detect the rising edge of a TTL signal as a spike. So I want the ADC channels work in uV mode and the electrode channels work in STD mode.

Thanks a lot for building this powerful open source system.

yaknerv, 2022-08-16:
BTW, the gui version is 0.6.0

jsiegle, 2022-08-16:
Thanks for reporting this! It looks like the display is failing to update for channels >19, even though the underlying threshold is actually changing. For example, in the Spike Viewer you can see the detection threshold (orange line) moving when you change the threshold, and the new threshold value is saved in the settings XML file. I'm not sure what's causing this glitch yet, but we'll figure out how to fix it ASAP.

It is possible to configure electrodes to use ADC channels as input, but it sounds like it might be better to use the [Crossing Detector](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Crossing-Detector.html) plugin for this application. That will create a TTL event whenever a rising edge is detected.

anjaldoshi, 2022-09-22:
Fixed in [v0.6.1](https://github.com/open-ephys/plugin-GUI/releases/tag/v0.6.1)

closed 2022-09-22 by anjaldoshi