open-ephys/plugin-GUI #520: Duplicated signal channels

skim0119, 2022-08-23:
Thank you for the great work always! We encountered some bug as we updated the GUI version to `0.6+`.

**Description**

We are operating two 64-channel headstages, and we have noticed the _recording from the first 64 channel LFP view is the exact copy of the following 64 channels_. The recorded data also showed duplication. We were able to record all 128 channels when we used the GUI version `0.5.5.5` and `0.5.4`and the issue only appeared when we upgraded the version to `0.6.0` and `0.6.1`. 

**Things we did to debug**

We couldn't resolve the issue trying:

- Clean-removed the software (including configuration files) and reinstall.
- Tried on Mac and Windows.
- Tried both `0.6.0` and `0.6.1`

**Other**

- We reverted the version back to `0.5.5.5` and the above issue was fixed.
- We have not seen any issue with two 32-channel headstages (64 channels total).

Do we have any other report with a similar bug, or is there any configuration that might have caused this issue?

jsiegle, 2022-08-23:
What version of the Rhythm Plugins do you have installed? The version number is visible via the Plugin Installer. We fixed a similar bug in version 0.2.1, so please upgrade to the latest version if you haven't done so yet.

Also, are you using two cables, or one cable plus splitter?

skim0119, 2022-08-26:
We confirmed that the problem was fixed after upgrading `Rhythm Plugins` to `0.2.1`. Thank you for quick support!

closed 2022-08-26 by skim0119