open-ephys/plugin-GUI #521: Channel mapping issue

rjp109, 2022-08-24:
Hello,

I have been using an EIB 64 v1.3 Omnetics flexdrive with one Intan RHD2132 32-channel headstage. 
Excluding the ref/gnd, I used 7 gold pins on the EIB (1.1,1.3,5.1,5.3,7.3,8.1,8.3 - I took this as 1.1 being top left, 1.3 being bottom left of the four pinouts). Now, on openephys LFP viewer i clearly see signal in 7 channels (2,6,8,13,15,25,31) with noise in the rest. However, when i try to map this using the provided EIB channel maps, it does not make sense for where i am seeing signal and which pins i have used. I have tried all sorts of inverting the map etc but still nothing makes sense (sometimes about half the channels do correspond, but never all 7).

Could you please give some advice on this?

Presumably pin in0 on the headstage equals channel 1 on the LFP viewer?

Many thanks

jsiegle, 2022-08-26:
@ahleighton or @ChucklesOnGitHub, can you help with this one?

ahleighton, 2022-08-29:
Hi @rjp109, thanks for pointing that out! We checked out the channel map and indeed found an error. We've updated it [here](https://docs.google.com/spreadsheets/d/1WYDymxNqGRtFPxn69H9JzeMgePpXcFSPHiWJYBE0lu4/edit#gid=1425490032), could you check that it now matches what you see?

rjp109, 2022-08-31:
Hello,

Yes great that now makes sense! Thank you for updating.

closed 2022-09-26 by anjaldoshi