open-ephys/plugin-GUI #522: network events plugin

neurochen, 2022-09-07:
I wonder if anyone encountered similar issues for GUI 0.6.x and the network events plugin in Windows 10 . After adding this plugin into the pipeline, we can't start acquisition by either manually pressing the start button or sending TCP commands via this plugin. It used to work perfectly for 0.5.x.

We notice that this plugin is now installed as a filter node, which is different from the older version for the source node. The only way we can get it working as before is to recompile the plugin as a source node. But it's still imperfect, as most of the time it didn't start recordings as usual. 

So, I am wondering if there is a template configuration file with the network events plugin included. That would be super helpful. Thanks.

jsiegle, 2022-09-08:
The latest version of the Network Events plugin works a little differently...in v0.6.x there's a strict requirement that all events must be associated with a data stream, which is a collection of synchronously sampled continuous channels. Because the Network Events plugin doesn't generate any continuous data on its own, it must be inserted into a signal chain that already has at least one incoming data stream – hence the need for it to be a filter, rather than a source. Events received over the network connection will be added to all streams passing through the plugin, with timestamps that are relative to each stream's local clock.

If you add the Network Events plugin after a source node, are you still unable to start acquisition?

For remote control functionality like starting/stopping acquisition, we recommend using the GUI's built-in [HTTP Server](https://open-ephys.github.io/gui-docs/User-Manual/Remote-control.html), since it doesn't depend on a plugin. But you can continue to use the Network Events plugin for this if you already have code written to interact with it.

neurochen, 2022-09-09:
Thanks, Josh. Your comments resolved my issues. The trick would be to 1) put the network events right after the source node, and 2) rebuild the entire pipeline from scratch, even if it's a migration from v0.6.0 to v0.6.1. Right now, everything works as expected with the standard plugin. 

We would like to stick to the network events plugin for starting/stopping acquisition, as our code is there already. But I do like the new RESTful API, with so much flexibility to specify recordings.

closed 2022-09-09 by neurochen