open-ephys/plugin-GUI #524: v0.6.0 Recording a custom data thread plugin without event channels makes the application crash

simonoxen, 2022-09-22:
Hi, I'm developing a [data thread plugin that streams data from the NeuroOmega device](https://github.com/netstim/OpenEphysNeuroOmega).

I realized that when I wanted to record with Record Node, the application crashes. The plugin does not create any event channels and the record node calls `eventChannels.getLast()` [here](https://github.com/open-ephys/plugin-GUI/blob/main/Source/Processors/RecordNode/RecordNode.cpp#L447). This makes the GUI crash.

Thanks, let me know if you're interested in me submitting a PR fixing this issue.

Simon

anjaldoshi, 2022-09-22:
Hi,

Data Thread plugins are supposed to create at least one event channel . That's why Record Node expects at least one event channel to exist in the data that it's receiving. If your external device isn't generating TTL events to add to the event channel, you can still create one and keep it unused. Is there a reason you'd like to not create any event channel in your Data Thread plugin?

-Anjal

simonoxen, 2022-09-23:
Hi, yes, that's what I ended up doing. I now create an event channel and don't use it. (This solution is however a bit hacky. I'd say it's better practice to fix underlying issue.)

I just reported this for future developers not to encounter the same issue.

Thanks!

anjaldoshi, 2022-09-23:
I agree that it's a bit hacky in your case, but I'd say, even if the data thread is not generating events currently, it's important to have an event channel to be able to take advantage of the Record Node's automated synchronization abilities in the future. Especially since data threads are designed to bring data into the signal chain from external hardware with different clocks.

closed 2022-09-26 by simonoxen