open-ephys/plugin-GUI #525: No output to console

rhayman, 2022-09-23:
Hi all,
I've compiled both "Release" and "Debug" builds of OE 0.6.1 on Windows 10 using Visual Studio 2019. Under both builds there is no output to the console beyond an initial line starting with "Session Start Time:" then the current date + time. I can see the cursor on the left hand side of the console scrolling down the page but no text output.

I am currently updating a plugin which is causing OE to crash (with the resulting message window saying OE has stopped working) however the activity log file is empty with a size of 0KB.

anjaldoshi, 2022-09-23:
Hi,

Did this start happening the first time you compiled v0.6.1 of the GUI or did it happen after you built and installed the plugin that you're upgrading? Can you delete everything from the `plugins-GUI\Build` directory and then run CMake and compile the GUI again? 

Also, what plugin are you trying to upgrade? I can help you debug the crash if I can get access to the code. Sometimes the GUI doesn't log any messages if it doesn't get time to launch properly before crashing.

rhayman, 2022-09-24:
Hey Anjal, thanks for the prompt reply.

Ignoring the plugin for now - I deleted the contents of the Build folder, pulled the latest commits (so now OE vers 0.6.2), ran CMake and compiled again. The problem persists on both Debug and Release under Windows 10.

I performed the same steps on Linux (pop-os so basically Ubuntu) and output to console is fine (Debug build) so this has to be a Windows issue. I have another machine in my lab with Windows but will only be able to access that on Monday so will provide an update then.

rhayman, 2022-09-26:
Well the problem is fixed although I'm not quite sure how. My other machine had a slightly later version of CMake (3.19) compared to the machine with the problem (which had CMake 3.16.2). I uninstalled CMake and installed the latest version (3.23.3), deleted the local clone of the plugin-GUI repo and re-cloned. Rebuilt and Debug build is now outputting to console. Marking this as closed and me as confused. Thanks for help again

closed 2022-09-26 by rhayman